Pilloxa/react-native-nordic-dfu #10: error: method does not override or implement a method from a supertype

jason2li, 2017-10-13:
I'm am trying to get this up and running, but once I install this plugin, I can no longer run the application.  I get the following error:

```
node_modules\react-native-nordic-dfu\android\src\main\java\com\pilloxa\dfu\RNNordicDfuPackage.java:19: error: method does not override or implement a method from a supertype
    @Override
    ^
1 error
:react-native-nordic-dfu:compileReleaseJavaWithJavac FAILED
```

I am working in Windows 10 and testing on Android (IOS will be next).

vikeri, 2017-10-16:
This is probably an issue with version of RN greater than 0.47, this is the fix that was done in another library: https://github.com/vikeri/react-native-background-job/pull/46/files

I'll try to deploy it ASAP

vikeri, 2017-10-16:
If anyone wants to test it and then submit a PR I'd happily merge it 👍

dariuszseweryn, 2018-05-19:
Does any of the above PRs qualify for merging?

PR #20 (2018-05-22): Remove @Override annotation in RNNordicDfuPackage
Since RN@0.47 (see https://github.com/facebook/react-native/commit/ce6fb337a146e6f261f2afb564aa19363774a7a8) the `@Override` annotation is no longer welcomed by the compiler as the function was removed.

And please make a release with this change if possible 👍
Fixes #10
closed 2018-05-22 by Looveh