Pilloxa/react-native-nordic-dfu #15: iOS error when building

stinodes, 2017-11-23:
After linking the library and trying to build the iOS app, I get the following error:
```
ld: /Users/stinodes/projects/com.minze.minzeflow/ios/Zip.framework/Zip compiled with newer version of Swift language (4.0) than previous files (3.0) file  '/Users/stinodes/projects/com.minze.minzeflow/ios/Zip.framework/Zip' for architecture armv7
```
I am sadly not an experienced iOS developer. To me an easy fix seems to be to install a Swift 3.0 version of Zip, but I might be mistaken of course.  
If you guys have a version that has Swift 3.0 frameworks, then that'd be great! If not, would an older version of `Zip` work well? (assuming that swapping out the swift 4.0 code even fixes it)

Looveh, 2017-11-23:
Version 1.1.0 of this library still had support for Swift 3, I suggest trying it out with that version :)

Looveh, 2017-11-23:
The reason we upgraded to only support Swift 4 was due to changes made in iOS 11 that made this library's dependencies unable to support Swift 3

stinodes, 2017-11-23:
I had just installed that, assuming that was the case. After cleaning and relinking, finally got it to build! Thanks for the fast response!

stinodes, 2017-11-24:
I'm still not getting my app to build, due to another issue.
Instead of [react-native-ble-manager](https://github.com/innoveit/react-native-ble-manager), I am using [react-native-ble-plx](https://github.com/Polidea/react-native-ble-plx).  
Linking this library in iOS results in the following error:
```
Undefined symbols for architecture armv7:
  "_OBJC_CLASS_$__TtC16BleClientManager16BleClientManager", referenced from:
      objc-class-ref in libBleClient.a(BleClient.o)
  "_OBJC_CLASS_$__TtC16BleClientManager8BleEvent", referenced from:
      objc-class-ref in libBleClient.a(BleClient.o)
ld: symbol(s) not found for architecture armv7
```
I have no idea how adding this library would result in an error somewhere seemingly unrelated. Maybe you guys have an idea? 😅 
  
Thanks again!

neomegas, 2018-04-16:
Dear, I also use react-native-ble-plx instead of your react-native-ble-manager.
From your explanation, I've struggled to understand step 8, which is    

       Call [RNNordicDfu setCentralManagerGetter:<...>] with a block argument that returns 
       your CBCentralManager instance (see example project for how this may be done). 
       It is assumed that you have initiated this instance outside of this library, 
       for flexibility.

as I try to implement with plx. I need to change header file and add other function. If possible, can you give me some clue.

I appreciate in advance!

Looveh, 2018-04-17:
So in iOS all BT central operations ultimately go through an instance of the class [CBCentralManager](https://developer.apple.com/documentation/corebluetooth/cbcentralmanager). For this library to function it needs a pointer to the instance. I haven't worked with `react-native-ble-plx` myself but under the hood it must be referencing an instance of this class.

Unfortunately I don't have the time or resources to dig deep into your issues but I'm certain that if you manage to extract a pointer to the `CBCentralManager` instance from `react-native-ble-plx` you will be able to pass it to this lib and progress 😊

neomegas, 2018-04-17:
Thank you for your comment. Now I can go further. Only thing is I'm too novice at swift.😊 Good news is it works on android. Definitely, I say you did fabulous job. Even though I might take dfu from ios at this time, I do try again after getting some ios knowledge. Thanks again!

vikeri, 2019-02-15:
Closing due to no activity

closed 2019-02-15 by vikeri