Pilloxa/react-native-nordic-dfu #17: update firmware not working got error on iOS

avneesh30, 2018-03-09:
when start NordicDFU.startDFU for update firmware got error.
**Error: Attempted to start DFU without central manager getter**
i have use react-native-ble-plx  v0.8.0 and react-native-nordic-dfu v 1.1.2

code - 
```
 NordicDFU.startDFU({
                            deviceAddress: obj.MacAddress,
                            name: obj.DeviceName,
                            filePath: filePath
                        })
                            .then(res => console.log(""))
                            .catch(console.log);
```
filePath is 

> /var/mobile/Containers/Data/Application/6DD7EC4C-47CB-486D-B132-A39F6813D474/Documents/CoolRIntelligence/CSS-0201.2.10.zip

obj.MacAddress is 

> DFE45376-F6A1-BBEF-02F5-3AH8C0342FE1

obj.DeviceName is 

> CSS-020100080651

[image]

Looveh, 2018-03-20:
You'll need to pass a block to the lib that returns an initialized CBCentralManager instance, see [here](https://github.com/Pilloxa/react-native-nordic-dfu/blob/master/example/ios/NordicDFUExample/AppDelegate.m#L37) for how it's done in the example project.

closed 2018-03-20 by Looveh

avneesh30, 2018-11-19:
this property is not in react-native-ble-plx 1.0.0 - CBCentralMan   @
anyone can help me to resolve this issue. @

urielges, 2019-03-21:
Hi, did you resolve this issue?

ThomasStubbe, 2019-09-06:
@avneesh30 I'm having the same problem: trying to combine this library with react-native-ble-plx... Did you manage to get it working?

rk85097, 2019-12-22:
Any news on this? Same problem here...

rk85097, 2019-12-22:
I might have found a solution for ble-plx lib.

In you app delegate didFinishLaunchingWithOptions add:
`  [[NSNotificationCenter defaultCenter] addObserver:self selector:@selector(setNordicCentralManagerGetter:) name:@"ble_manager" object:nil];
`

Then add the following method to your app delegate:

`- (void)setNordicCentralManagerGetter:(NSNotification *)notification
{
  NSDictionary *userInfo = notification.userInfo;
  CBCentralManager *manager = [userInfo objectForKey:@"manager"];
  [RNNordicDfu setCentralManagerGetter:^() {
    return manager;
  }];
}`

Then go to file _node_modules/react-native-ble-plx/ios/BleClientManager/BleModule.swift_ and add in line 83:

`let managerDict:[String: CBCentralManager] = ["manager": manager.manager]
        NotificationCenter.default.post(name: NSNotification.Name(rawValue: "ble_manager"), object: nil, userInfo: managerDict)`

williamgoodhew, 2020-10-16:
@rk85097 what version of ble-plx were you using? I can't find the BLeClientManager folder in the Node Module. I'm on 2.0.1

niclaslindberg, 2023-10-17:
Hi @williamgoodhew 
I'm stuck at the exact same problem. Very kind if you could share any success even though it was some time ago. :) 
I'm using:
`react-native-nordic-dfu=3.2.1` 
`react-native-ble-plx=2.0.3`
Getting the error: `Error: Attempted to start DFU without central manager getter`
Can it be done with  `ble-plx` or do I need to shift ble lib?
Super thankful for any feedback.

dev-simpleplan, 2023-11-07:
> Hi @williamgoodhew I'm stuck at the exact same problem. Very kind if you could share any success even though it was some time ago. :) I'm using: `react-native-nordic-dfu=3.2.1` [...]

Hi, i am facing the same issue. have you found any solution for this issue?