Pilloxa/react-native-nordic-dfu #24: Issue with building

StrawHatAaron, 2018-07-23:
Hello I am running:

> Mac OS High Sierra 10.13.6
> xCode 9.4.1
>
> react-native-cli: 2.0.1
> react-native: 0.56.0
>
> npm 6.2.0
> yarn 1.7.0
> watchman 4.9.0

My team and I are switching over to React Native however building for iOS is giving us some issues. We get the following 3 errors and have been trying to solve it for about 3 days now.

```
Undefined symbols for architecture x86_64:
  "_OBJC_CLASS_$__TtC13iOSDFULibrary11DFUFirmware", referenced from:
      objc-class-ref in libRNNordicDfu.a(RNNordicDfu.o)
  "_OBJC_CLASS_$__TtC13iOSDFULibrary19DFUServiceInitiator", referenced from:
      objc-class-ref in libRNNordicDfu.a(RNNordicDfu.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

Any help would be very appreciated!

StrawHatAaron, 2018-07-23:
[image]

tried manual linking. 
deleting derived data.
clearing react native cache.
ect.

I am really stuck in a road block here :/

StrawHatAaron, 2018-07-23:
Hey @vikeri thank you for your quick response! I appreciate it. I downgraded to 55 and followed your manual install instructions. I am now getting it to build. 

Thanks so much for all your great work👍

vikeri, 2018-07-26:
As soon as we get our PR merged in the parent lib here: https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/pull/203 we'll release a new version to npm

VitamineC, 2018-07-27:
Good day. 
Get same with @StrawHatAaron  error: 
```
dyld: Library not loaded: @rpath/libswiftCore.dylib
  Referenced from: /private/var/containers/Bundle/Application/3655EA0D-A97E-4EC4-B100-E67B0106C3FE/ROMBOT.app/Frameworks/iOSDFULibrary.framework/iOSDFULibrary
  Reason: image not found

```

RN: 0.55

I will be grateful for any help

vikeri, 2018-07-27:
@VitamineC did you try this: https://github.com/Pilloxa/react-native-nordic-dfu/issues/24#issuecomment-406965309

Looveh, 2018-07-30:
@StrawHatAaron @VitamineC You're most likely running into that issue due to your Xcode/iOS project compiling with a different Swift version than the one that the embedded libraries of this lib was compiled with. We ran into this issue ourselves when upgrading Xcode from 9.2 to 9.4. As @vikeri said using this branch `Pilloxa/react-native-nordic-dfu#swift-4.1` should solve the problem.

closed 2018-07-31 by Looveh