Pilloxa/react-native-nordic-dfu #25: Dfu service not found

1058278168, 2018-08-01:
DFU state: DFU_FAILED
BleTest.js:99 upgrade failed ===  Error: DFU SERVICE NOT FOUND
    at createErrorFromErrorData (NativeModules.js:121)
    at NativeModules.js:78
    at MessageQueue.__invokeCallback (MessageQueue.js:398)
    at MessageQueue.js:137
    at MessageQueue.__guardSafe (MessageQueue.js:314)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (MessageQueue.js:136)
    at debuggerWorker.js:70

vikeri, 2018-11-28:
What version of Android are you targeting?

Yphfeng, 2018-12-24:
I also have this problem, my Android version is 6.0.1

Manish123Pandia, 2019-02-14:
facing same issue on  IOS 12.1

vikeri, 2019-02-15:
Are you sure that you have flashed the bootloader on the device?

Looveh, 2019-04-16:
Closing due to inactivity

closed 2019-04-16 by Looveh