Pilloxa/react-native-nordic-dfu #26: App crashes when running on Xcode 10

SchwarzerSchnee, 2018-08-02:
Hello. Had to do lots of upgrades recently, the Swift 4.1 issue ( 
https://github.com/Pilloxa/react-native-nordic-dfu/issues/24 ) being one of them, and at least that eventually worked...somehow. Still, one different problem remains:

Among the iPhones I'm working with, one is on iOS 12 beta, which requires Xcode 10 beta for developing. After following all the manual installation instructions, Carthage updating and whatnot, my project builds, but then running it causes the following error to occur on startup:

> dyld: Symbol not found: __T0BoWV
>   Referenced from: /private/var/containers/Bundle/Application/E5D7F3EC-6EEA-4A07-87E2-FAA7FD91AD94/_______.app/Frameworks/Zip.framework/Zip
>   Expected in: /private/var/containers/Bundle/Application/E5D7F3EC-6EEA-4A07-87E2-FAA7FD91AD94/_______.app/Frameworks/libswiftCore.dylib
>  in /private/var/containers/Bundle/Application/E5D7F3EC-6EEA-4A07-87E2-FAA7FD91AD94/_______.app/Frameworks/Zip.framework/Zip

Building the app via Xcode 9.4 and installing it via Testflight on the iOS 12 device works fine, but can only run it via Xcode on iOS 11 and lower. Any fixes soon, or could there be something additional I have to do myself?

react-native-nordic-dfu: swift-4.1
react-native: 0.56
react-native-ble-plx: 0.10
iOS: 12.0
Xcode: 10.0 beta 4

Looveh, 2018-08-02:
Thanks for the info @SchwarzerSchnee. We have no plan yet to add support for Xcode 10, although it's an issue we need to tackle at some point in time, so if it's urgent for you I suggest not waiting for a fix. 

On the face of it, it sounds to me like the Zip dependency (over at https://github.com/marmelroy/Zip) might not support Xcode 10, so I'd start by looking there.

arthare, 2018-09-28:
Chiming in to say I managed to get past this by cloning marmelroy/Zip, building it myself with the swift 4.2 libraries, and plugging it into my forked react-native-nordic-dfu.  Successfully got past the dyld __TB0BOWV issue for Zip.framework, and then it came up again iOSDFULibrary.framework.  Going to try to start the adventure again for that...

vikeri, 2018-10-01:
@arthare Please submit a PR and we'll merge it 👍

arthare, 2018-10-02:
It's imperfect: I'm not good enough at wrangling XCode to produce release-mode `Zip.framework` and `iOSDFULibrary.framework` that included bitcode for all the architectures (specifically, it was missing for arm64), which meant I had to turn off bitcode for my app in order to get it onto the store.  Not a problem for me, but it'd be a regression for react-native-nordic-dfu I think.  I wouldn't be very confident submitting a binary that I built at all.

That said, our metrics show that last Friday's build has been used to flash devices in the wild, so it wasn't all bad :-)

Looveh, 2019-04-16:
This should be solved with v3.0.0 when using CocoaPods

closed 2019-04-16 by Looveh