Pilloxa/react-native-nordic-dfu #30: Only Upload 1 part

pyramid-ali, 2018-09-13:
Hi,
when I use this library to update our device firmware only application layer part will be uploaded. but when I use NRF toolbox  all parts (bootloader + application) will be uploaded>
with dfu progress listener I only see 1 part, is there any bug?
should I extract it to 2 parts for uploading?

vikeri, 2018-09-13:
Hi!
We currently don't support uploading both bootloader and application at the same time. This branch has experimental support for uploading the bootloader:
https://github.com/Pilloxa/react-native-nordic-dfu/tree/fw-type

vikeri, 2018-09-13:
You can update softDevice + bootloader + application at the same time if you use that branch