Pilloxa/react-native-nordic-dfu #31: Firmare not specified

MD3XTER, 2018-09-13:
The second I call  `NordicDFU.startDFU` It gives me this error:

[image]

This is my code:
```
NordicDFU.startDFU({
    deviceAddress: peripheralID,
    name: "My Device",
    filePath: firmwarePath
})
  .then(res => console.log("Transfer done:", res))
  .catch(error => console.log(error));
```

This is the path to the firmware:
`
/var/mobile/Containers/Data/Application/5869AB60-ADF7-4FBD-94BD-2C9C2A10D626/Documents/RNFetchBlob_tmp/RNFetchBlobTmp_6k32nkgwxwrx77j66ofrsk.zip
`

My guess is that it has to do something with permissions, but I am not sure it's necessary.

p.s I get this error on iOS, haven't checked on Android yet.

@Looveh @vikeri thanks for this awesome library!!!

vikeri, 2018-09-14:
We have the following settings for RNFetchBlob:
```js
{fileCache: true,
 appendExt: "zip"}
```

Then we get `.path` from the returned RNFetchBlob "blob".

Other than that your code looks good. Can't remember if there are any other gotchas. The error thrown is from the original lib from Nordic.

MD3XTER, 2018-09-14:
This is the code that I use for RNFetchBlob:
```
const response = await RNFetchBlob.config({
      fileCache: true,
      appendExt: "zip"
    })
      .fetch(
        "GET",
        url,
        JSON.stringify({
          app,
          version
        })
      )
      .progress((received, total) => onProgress(received / total));
    return response.path();
```

I am really confused by the error, I am not sure what does it means...

p.s How did you implement the OTA server? I am using [auto-update-server](https://github.com/Automattic/auto-update-server), but I think there are other solutions on the web

vikeri, 2018-09-14:
We're acutally using an experimental branch, so try: Pilloxa/react-native-nordic-dfu#swift-4.1

MD3XTER, 2018-09-14:
@vikeri Yep, that's what I am using...

vikeri, 2018-09-14:
Weird, does the example project work for you?
Regarding our CI flow we build the FW package on our CI server and then we upload it to S3. Then the app checks with our API if there are any new versions available. If there is the API generates a signed link for the app to download and then the app updates the device.

MD3XTER, 2018-09-14:
@vikeri I tried the example project with: Pilloxa/react-native-nordic-dfu#swift-4.1 and it throws me the same error

MD3XTER, 2018-09-14:
@vikeri should I forward this issue to [IOS-Pods-DFU-Library](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library)?

MD3XTER, 2018-09-14:
@vikeri It wasn't a bug with any library, Instead of downloading the actual firmware, I was downloading the `HTML` webpage and adding `.zip` extension :D

I got confused because on the browser if I access that link it downloads the firmware automatically, but I forgot that with `RNFetchBlob` I have to specify the exact path.

Sorry for taking your time, I hope this issue will be somehow helpful for other people, but I would also like the error to be more descriptive instead of just `firmware not specified`, maybe `specified file doesn't contain any hex files` or smth like that, I will see maybe propose a feature request from the [IOS-Pods-DFU-Library](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library)

closed 2018-09-14 by MD3XTER

vikeri, 2018-09-17:
No problem! Happy that it got resolved! Thanks! Your contributions are welcome 😄

daihieptn97, 2020-06-18:
I update from ^3.0.1 to ^3.2.0 and also got this error

thilosuelberg, 2021-01-10:
For iOS, the filePath must be a URL. If you want to load a local file, prefix the absolute path with "file://". Maybe that is your problem. Good luck!