Pilloxa/react-native-nordic-dfu #32: "RNNordicDfu.h" file not found

pyramid-ali, 2018-09-17:
when I clone this repo and run ios version, I get this error "RNNordicDfu.h" file not found
is there any solution to it?

vikeri, 2018-09-18:
Wasn't involved in the iOS code so I'll let @Looveh answer that when he has time

MD3XTER, 2018-09-18:
I had the same problem, the solution for me was to link the library manually again, see [ios-manual-linking](https://github.com/Pilloxa/react-native-nordic-dfu#manual-installation)

MD3XTER, 2018-09-22:
@pyramid-ali 

Sure, this is my project settings, It's the same as described here: [ios-manual-linking](https://github.com/Pilloxa/react-native-nordic-dfu#manual-installation)

_Take in mind I also have other stuff besides `react-native-nordic-dfu`_ installed.

**Framework Search Paths**
[image]

**Always Embed Swift Standard Libraries**
[image]

**Embedded binaries & Linked Frameworks and Libraries**
[image]

**Project Files**
[image]

**AppDelegate.m** 
```
#import "AppDelegate.h"
#import <AppCenterReactNativeCrashes/AppCenterReactNativeCrashes.h>
#import <AppCenterReactNativeAnalytics/AppCenterReactNativeAnalytics.h>
#import <AppCenterReactNative/AppCenterReactNative.h>
#import <Firebase.h>
#import "RNNordicDfu.h"
#import "BleManager.h"

#import <React/RCTBundleURLProvider.h>
#import <React/RCTRootView.h>

@implementation AppDelegate

- (BOOL)application:(UIApplication *)application didFinishLaunchingWithOptions:(NSDictionary *)launchOptions
{
  [FIRApp configure];
  
  NSURL *jsCodeLocation;

  [AppCenterReactNativeCrashes registerWithAutomaticProcessing];  // Initialize AppCenter crashes

  [AppCenterReactNativeAnalytics registerWithInitiallyEnabled:true];  // Initialize AppCenter analytics

  [AppCenterReactNative register];  // Initialize AppCenter 

  jsCodeLocation = [[RCTBundleURLProvider sharedSettings] jsBundleURLForBundleRoot:@"index" fallbackResource:nil];

  RCTRootView *rootView = [[RCTRootView alloc] initWithBundleURL:jsCodeLocation
                                                      moduleName:@"AppName"
                                               initialProperties:nil
                                                   launchOptions:launchOptions];
  rootView.backgroundColor = [[UIColor alloc] initWithRed:1.0f green:1.0f blue:1.0f alpha:1];

  self.window = [[UIWindow alloc] initWithFrame:[UIScreen mainScreen].bounds];
  UIViewController *rootViewController = [UIViewController new];
  rootViewController.view = rootView;
  self.window.rootViewController = rootViewController;
  [self.window makeKeyAndVisible];
  
  [RNNordicDfu setCentralManagerGetter:^() {
    return [BleManager getCentralManager];
  }];
  
  // Reset manager delegate since the Nordic DFU lib "steals" control over it
  [RNNordicDfu setOnDFUComplete:^() {
    NSLog(@"onDFUComplete");
    CBCentralManager * manager = [BleManager getCentralManager];
    manager.delegate = [BleManager getInstance];
  }];
  [RNNordicDfu setOnDFUError:^() {
    NSLog(@"onDFUError");
    CBCentralManager * manager = [BleManager getCentralManager];
    manager.delegate = [BleManager getInstance];
  }];
  
  return YES;
}
```

pyramid-ali, 2018-09-22:
I've done all of that, but I still have the issue. my project setting is same as you, but I can't understand why this error thrown for me.
anyway tnx a lot for your help. it seems I need a pro IOS developer to invest in my project to solve my issue

MD3XTER, 2018-09-22:
@pyramid-ali What version of the xCode and `react-native-nordic-dfu` do you use?

pyramid-ali, 2018-09-22:
xcode 9.4.1
react-native-nordic-dfu: I forked this repo to merge fw-type branch.

pyramid-ali, 2018-09-22:
swift-4.1 branch is included in my forked repo. I had same issue before but now It's ok.
my only problem is that I can't import header file of this library in AppDelegate to call needed function. it says that there is no such file. it's weird. I don't know what I have this problem. but I'm sure I followed all manual installation steps.

MD3XTER, 2018-09-22:
@pyramid-ali I also have struggled with it a lot, have reinstalled the project around 10 times and couldn't get it done correctly. I will suggest trying again from scratch without `fw-type`, just `swift-4.1` and follow the instructions carefully.

I know it's frustrating, been there, got to a point where I wanted to hire someone to just install this library :D

pyramid-ali, 2018-09-22:
I should do this. but my problem is I need fw-type branch. without it this library doesn't work for me. our device need to update all 3 parts of dfu. but I think in IOS only application part will be uploaded and it is unacceptable for my application.

MD3XTER, 2018-09-22:
@pyramid-ali I am not sure, but after reading the README of [IOS-Pods-DFU-Library](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/master/documentation.md#usage), I have an impression that if you pack the bootloader, application and soft device in one zip file, It may automatically update everything correctly.

But I am not sure at all about that, it's just my assumption.

pyramid-ali, 2018-09-23:
finally I found the problem. It's related to Header search path.
thanks a lot @MD3XTER  for your help

vikeri, 2018-09-24:
Is there anything we can fix in order to avoid this issue?

vikeri, 2018-10-01:
Since I merged your PR @pyramid-ali I'm closing this issue now.

closed 2018-10-01 by vikeri