Pilloxa/react-native-nordic-dfu #35: Exception when running startDfu in ios

pyramid-ali, 2018-10-07:
Hi, when I run `startDfu` command in ios I get this error:
```
Exception '*** -[__NSPlaceholderArray initWithObjects:count:]: attempt to insert nil object from objects[0]' was thrown while invoking startDFU on target RNNordicDfu with params (
    "E60161B8-7C2B-BB7C-1D60-EEE4BC3E6C1D:E60161B9",
    Ninix,
    "/var/mobile/Containers/Data/Application/45F13F0D-F4B6-4BD0-A19F-A201C3D4F2D5/Documents/RNFetchBlob_tmp/RNFetchBlobTmp_tuv5vjjvmmvd67t2lhlt.zip",
    3250,
    3251
)
```

I'm using:
`react-native`: 0.57
`react-native-nordic-dfu`: "swift4.1 branch"

closed 2018-10-10 by pyramid-ali

blakeparkinson, 2019-10-26:
@pyramid-ali Do you remember how you resolved this by any chance? I'm running into the same issue on my end. My suspicion is that we don't have access to the directory on the device where I'm downloading the file. I noticed that if I also just hard-code some nonsense path that doesn't exist, I get the same error.

Appreciate it if you could provide any guidance.

ezranbayantemur, 2020-06-08:
Same here. Any update?

blakeparkinson, 2021-11-09:
@franzejr I was ultimately able to get this working. It's been a while though. I had to dip down into the java code a fair amount