Pilloxa/react-native-nordic-dfu #39: iOS build fail

nvelozsavino, 2018-10-16:
I'm getting the following error when compiling for iOS
the Xcode version is 10, I'm using v1.2.1 of react-native-nordic-dfu
Undefined symbols for architecture arm64:
  "_OBJC_CLASS_$__TtC13iOSDFULibrary11DFUFirmware", referenced from:
      objc-class-ref in libRNNordicDfu.a(RNNordicDfu.o)

Any ideas?

reekris, 2018-10-16:
I had the same issue. After following the manual installation steps in the readme for iOS from step 4 onwards my project built successfully. 

Not sure if there is something is wrong with the link command or if it’s just unclear from the readme that you have to follow those steps as well to make it work.

nvelozsavino, 2018-10-18:
I manage to build and run the app, I did the following:

1. Execute carthage update --platform iOS from the node_modules/react-native-nordic-dfu/ios folder
2. Copy the folders Zip.framework and iOSDFULibrary.framework and the files Zip.framework.dSYM and   iOSDFULibrary.framework.dSYM generated in the folder Carthage/build to my project's ios/ folder
3. Added both Zip and iOSDFULibrary frameworks folders and .dSYM to my Frameworks folder in XCode
4. Added to the embedded binaries in the project settings the Zip.framework and iOSDFULibrary.framework folders and Zip and iOSDFULibrary files that are inside each .framework folders (the one copied to my project's ios folder)
5. Enable Option "Always Embed Swift Standard Libraries" in Project Settings

It builds successfully but now I'm having an issue when starting the dfu I get the rejection of the promise with the following reason:

{
  "framesToPop": 1,
  "code": "nil_central_manager_getter",
  "nativeStackIOS": [
    "0   UnlimitedSkate                      0x00000001006edbf8 RCTJSErrorFromCodeMessageAndNSError + 152",
    "1   UnlimitedSkate                      0x0000000100686bb0 __41-[RCTModuleMethod processMethodSignature]_block_invoke_2.218 + 176",
    "2   UnlimitedSkate                      0x00000001008ae654 -[RNNordicDfu startDFU:deviceName:filePath:resolver:rejecter:] + 352",
    "3   CoreFoundation                      0x00000001969af660 <redacted> + 144",
    "4   CoreFoundation                      0x000000019688b980 <redacted> + 292",
    "5   CoreFoundation                      0x000000019688c564 <redacted> + 60",
    "6   UnlimitedSkate                      0x00000001006887c4 -[RCTModuleMethod invokeWithBridge:module:arguments:] + 2064",
    "7   UnlimitedSkate                      0x0000000100737bf4 _ZN8facebook5reactL11invokeInnerEP9RCTBridgeP13RCTModuleDatajRKN5folly7dynamicE + 664",
    "8   UnlimitedSkate                      0x0000000100737778 _ZZN8facebook5react15RCTNativeModule6invokeEjON5folly7dynamicEiENK3$_0clEv + 132",
    "9   UnlimitedSkate                      0x00000001007376e8 ___ZN8facebook5react15RCTNativeModule6invokeEjON5folly7dynamicEi_block_invoke + 28",
    "10  libdispatch.dylib                   0x0000000100e2f840 _dispatch_call_block_and_release + 24",
    "11  libdispatch.dylib                   0x0000000100e30de4 _dispatch_client_callout + 16",
    "12  libdispatch.dylib                   0x0000000100e38e88 _dispatch_lane_serial_drain + 720",
    "13  libdispatch.dylib                   0x0000000100e39b7c _dispatch_lane_invoke + 460",
    "14  libdispatch.dylib                   0x0000000100e43c18 _dispatch_workloop_worker_thread + 1220",
    "15  libsystem_pthread.dylib             0x00000001965c20f0 _pthread_wqthread + 312",
    "16  libsystem_pthread.dylib             0x00000001965c4d00 start_wqthread + 4"
  ],
  "domain": "RCTErrorDomain",
  "userInfo": null
}

reekris, 2018-10-18:
Did you follow step 9 of the Manual installation steps for iOS? You have to call `setCentralManagerGetter` from your native code. Cheeck out the example project for how it can be done [here](https://github.com/Pilloxa/react-native-nordic-dfu/blob/18a105721ce27d70e0ba03f204cd40b4866f58e6/example/ios/NordicDFUExample/AppDelegate.m#L37-L39)

nvelozsavino, 2018-10-18:
Thanks, I added the missing code and it passed that point, however, I'm still getting an error:

The EventEmmiter gave me the following states:
CONNECTING
DFU_PROCESS_STARTING
ENABLING_DFU_MODE
DFU_FAILED

and then I capture the following error:

{
  "framesToPop": 1,
  "code": "DFUErrorWritingCharacteristicFailed",
  "nativeStackIOS": [
    "0   UnlimitedSkate                      0x0000000100b11bb8 RCTJSErrorFromCodeMessageAndNSError + 152",
    "1   UnlimitedSkate                      0x0000000100aaab70 __41-[RCTModuleMethod processMethodSignature]_block_invoke_2.218 + 176",
    "2   UnlimitedSkate                      0x0000000100cd2110 -[RNNordicDfu dfuError:didOccurWithMessage:] + 452",
    "3   iOSDFULibrary                       0x000000010157e1a4 $S13iOSDFULibrary15BaseDFUExecutorPAAE23peripheralDidDisconnectyyFyycfU_ + 188",
    "4   iOSDFULibrary                       0x000000010157df6c $SIeg_IeyB_TR + 36",
    "5   libdispatch.dylib                   0x0000000101873840 _dispatch_call_block_and_release + 24",
    "6   libdispatch.dylib                   0x0000000101874de4 _dispatch_client_callout + 16",
    "7   libdispatch.dylib                   0x0000000101882a94 _dispatch_main_queue_callback_4CF + 1360",
    "8   CoreFoundation                      0x00000001969361bc <redacted> + 12",
    "9   CoreFoundation                      0x0000000196931084 <redacted> + 1964",
    "10  CoreFoundation                      0x00000001969305b8 CFRunLoopRunSpecific + 436",
    "11  GraphicsServices                    0x0000000198ba4584 GSEventRunModal + 100",
    "12  UIKitCore                           0x00000001c37acbc8 UIApplicationMain + 212",
    "13  UnlimitedSkate                      0x0000000100a0e8a4 main + 124",
    "14  libdyld.dylib                       0x00000001963f0b94 <redacted> + 4"
  ],
  "domain": "RCTErrorDomain",
  "userInfo": null
}

This time it seems a problem when running the dfu protocol

any ideas?

nvelozsavino, 2018-10-22:
I got the logs from the ios, the problem seems to be a write characteristic error when trying setting bootloader name.

2018-10-22 10:47:29.967539+0200 UnlimitedSkate[5169:1339653] logWith: 5 message: 'Connected to Unlimited_Mesc'
2018-10-22 10:47:29.967746+0200 UnlimitedSkate[5169:1339653] logWith: 5 message: 'Services discovered'
2018-10-22 10:47:29.967833+0200 UnlimitedSkate[5169:1339653] logWith: 1 message: 'Starting Secure DFU...'
2018-10-22 10:47:29.968028+0200 UnlimitedSkate[5169:1339653] logWith: 5 message: 'Connected to Unlimited_Mesc'
2018-10-22 10:47:29.968149+0200 UnlimitedSkate[5169:1339653] logWith: 5 message: 'Services discovered'
2018-10-22 10:47:29.968233+0200 UnlimitedSkate[5169:1339653] logWith: 1 message: 'Secure DFU Service found'
2018-10-22 10:47:29.968337+0200 UnlimitedSkate[5169:1339653] logWith: 1 message: 'Discovering characteristics in DFU Service...'
2018-10-22 10:47:29.968494+0200 UnlimitedSkate[5169:1339653] logWith: 0 message: 'peripheral.discoverCharacteristics(nil, for: FE59)'
2018-10-22 10:47:29.970989+0200 UnlimitedSkate[5169:1338172] logWith: 5 message: 'DFU characteristics discovered'
2018-10-22 10:47:29.971378+0200 UnlimitedSkate[5169:1338172] logWith: 1 message: 'Enabling indications for 8EC90003-F315-4F60-9FB8-838830DAEA50...'
2018-10-22 10:47:29.971492+0200 UnlimitedSkate[5169:1338172] logWith: 0 message: 'peripheral.setNotifyValue(true, for: 8EC90003-F315-4F60-9FB8-838830DAEA50)'
2018-10-22 10:47:29.986692+0200 UnlimitedSkate[5169:1338172] logWith: 1 message: 'Indications enabled for 8EC90003-F315-4F60-9FB8-838830DAEA50'
2018-10-22 10:47:29.986849+0200 UnlimitedSkate[5169:1338172] logWith: 10 message: 'Buttonless DFU indications enabled'
2018-10-22 10:47:29.986935+0200 UnlimitedSkate[5169:1338172] logWith: 15 message: 'Application with buttonless update found'
2018-10-22 10:47:29.987098+0200 UnlimitedSkate[5169:1338172] logWith: 1 message: 'Trying setting bootloader name to Dfu16751'
2018-10-22 10:47:29.987232+0200 UnlimitedSkate[5169:1338172] logWith: 1 message: 'Writing to characteristic 8EC90003-F315-4F60-9FB8-838830DAEA50...'
2018-10-22 10:47:29.987500+0200 UnlimitedSkate[5169:1338172] logWith: 0 message: 'peripheral.writeValue(0x02084466753136373531, for: 8EC90003-F315-4F60-9FB8-838830DAEA50, type: .withResponse)'
2018-10-22 10:47:30.046209+0200 UnlimitedSkate[5169:1338172] logWith: 20 message: 'Writing to characteristic failed'
2018-10-22 10:47:30.046413+0200 UnlimitedSkate[5169:1338172] logWith: 20 message: 'Error 3: Writing is not permitted.'
2018-10-22 10:47:30.046649+0200 UnlimitedSkate[5169:1338172] logWith: 1 message: 'Disconnecting...'
2018-10-22 10:47:30.046730+0200 UnlimitedSkate[5169:1338172] logWith: 0 message: 'centralManager.cancelPeripheralConnection(peripheral)'
2018-10-22 10:47:30.048176+0200 UnlimitedSkate[5169:1338172] logWith: 0 message: '[Callback] Central Manager did disconnect peripheral'
2018-10-22 10:47:30.048321+0200 UnlimitedSkate[5169:1338172] logWith: 5 message: 'Disconnected'
2018-10-22 10:47:30.048554+0200 UnlimitedSkate[5169:1338172] onDFUError