Pilloxa/react-native-nordic-dfu #42: Firmare not specified

ljubinkovicd-symphony, 2018-11-28:
As soon as I call `NordicDFU.startDFU` I get **Firmare not specified** error...

Not sure what I am doing wrong here:

```
const FB = RNFetchBlob.config({
  fileCache: true,
  appendExt: "zip"
});

FB.fetch("GET", "http://192.168.15.52.xip.io:8081/debugger-ui/myFirmwareFile.zip")
      .then(res => {
        console.log("file saved to", res.path());
        this.setState({ imagefile: res.path() });
      })
      .catch(error => {
        console.log(`FB_FETCH_ERROR: ${error}`);
      });

NordicDFU.startDFU({
      deviceAddress: peripheral.id,
      name: peripheral.name,
      filePath: this.state.imagefile
    })
      .then(res => console.log("Transfer done:", res))
      .catch(error => console.log(`Transfer error: ${error}`));
```

I've tried using the hex file as well, but no luck... Could you please explain what I am doing wrong here?

Thanks!

MD3XTER, 2018-11-28:
Hi, I am not sure what is the problem, but make sure you zip contains the right files. I had a similar issue but it was my fault, maybe it helps #31

ljubinkovicd-symphony, 2018-11-28:
@MD3XTER I've looked at your issue, but I am not downloading the HTML page. 

I literally do the following:

1. Connect the hardware to my laptop via usb
2. In firmware repo for that hardware, I run **make** command which builds the **.zip** file
3. I put that **.zip** file inside the root of my project, and the call `NordicDFU.startDFU`

So there should be nothing wrong with the **.zip** folder...

MD3XTER, 2018-11-28:
I am not sure what is the problem, but just double check that **.zip** contains the right files.

This is how my **.zip** looks:

[image]

```json
// manifest.json

{
    "manifest": {
        "application": {
            "bin_file": "application.bin",
            "dat_file": "application.dat"
        }
    }
}
```

Also, make sure you can access the **.zip** from that path.

ljubinkovicd-symphony, 2018-11-28:
Ok, I figured it out... for some reason it was telling me that the file was saved to:

```
/var/mobile/Containers/Data/Application/DD17EDE3-4DA3-4A07-9173-AA2B9073F037/Documents/RNFetchBlob_tmp/RNFetchBlobTmp_mwgdau6fiwdxlopjjx9wr.zip
```

but it wasn't..

So I just fired up a simple http server from the root directory: `python -m SimpleHTTPServer 3000`
and put the **.zip** folder in the root directory, and then I changed this line of code:

`FB.fetch("GET", "http://192.168.123:3000/cadence-v2.9-1-gedc1a56-app.zip")`

and that worked...

@MD3XTER Thanks for the help :)

RideShi, 2019-04-15:
Firmware not specified  
/var/mobile/Containers/Data/Application/9BA61605-2A61-4D85-8068-CCDECB1DE4EC/Documents/PluginStorage/com.chuangmi.newlock.ios/DFU/cm_app_dfu_package.zip

Looveh, 2019-04-16:
This error means that you've specified a path that isn't accessible from your app, please check where the file is stored since this shouldn't be a problem with this library

closed 2019-04-16 by Looveh

daihieptn97, 2020-06-18:
I update from ^3.0.1 to ^3.2.0 and also got this error

RRMoelker, 2020-08-20:
I got this error when the path did not include `file:///` on iOS. It did not contain that prefix when using react-native-blob fetch. Adding that fixed it for me.