Pilloxa/react-native-nordic-dfu #44: App store release issue. Code signing "Zip.framework" failed.

folofse, 2018-12-06:
I ran in to this problem when I tried to release my app to Appstore. 

Code signing "Zip.framework" failed.

I have followed and double checked the iOS linking guide, but can't get it to work.

VitamineC, 2018-12-17:
Same issue

hokasan, 2018-12-18:
Same issue here

dantetwc, 2019-01-08:
It is because the frameworks come with the package do not generate bitcode. I have recompiled the frameworks from source with bitcode generation enabled. You can download at https://drive.google.com/file/d/1hu7DB0KQkhUgRTb99CgMVxqt6yRHc1Qf/view?usp=sharing
Replace the framework inside the node_modules folder and everything should be fine

VitamineC, 2019-01-09:
@dantetwc 
Hello, I tried using your solution (install your PR), however I got a problem 
`ld: warning: Could not find auto-linked framework 'iOSDFULibrary'
Undefined symbols for architecture x86_64:
  "_OBJC_CLASS_$__TtC13iOSDFULibrary11DFUFirmware", referenced from:
      objc-class-ref in libRNNordicDfu.a(RNNordicDfu.o)
  "_OBJC_CLASS_$__TtC13iOSDFULibrary19DFUServiceInitiator", referenced from:
      objc-class-ref in libRNNordicDfu.a(RNNordicDfu.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)`

Probably u know how resolve this?

hokasan, 2019-01-09:
I get this error
`Undefined symbols for architecture x86_64:
  "_OBJC_CLASS_$__TtC13iOSDFULibrary11DFUFirmware", referenced from:
      objc-class-ref in libRNNordicDfu.a(RNNordicDfu.o)
  "_OBJC_CLASS_$__TtC13iOSDFULibrary19DFUServiceInitiator", referenced from:
      objc-class-ref in libRNNordicDfu.a(RNNordicDfu.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)`

VitamineC, 2019-01-09:
@hokasan  you can resolve this error and get my =) if you remove framework files from "Linked Frameworks and Libraries"

hokasan, 2019-01-09:
@VitamineC I now get the same error as you.....yaaaee.. :)

dantetwc, 2019-01-09:
@hokasan @VitamineC  are you guys running it on simulator or real device?

VitamineC, 2019-01-09:
@dantetwc  simulator, xcode x, deprecated build system

VitamineC, 2019-01-09:
@dantetwc, also I try change frameworks in node_modules  (using your google drive) and get same error

hokasan, 2019-01-09:
@dantetwc I was trying to use simulator as well.
I just tried to create an archive for appstore and that worked!

VitamineC, 2019-01-09:
@hokasan, 
however can you start your app from testfly?

dantetwc, 2019-01-09:
@hokasan @VitamineC please use the latest version of the frameworks here: https://github.com/dantetwc/react-native-nordic-dfu/tree/master/ios

I have updated the frameworks which also includes i386 and x86_64 arch used for simulator now

dantetwc, 2019-01-09:
However, if the frameworks are bundled with simulator arch, you will get rejected when you upload to AppStore or TestFlight. Add the following code to build script so that unused arch (ie: x86_64, i386) will be remove from the frameworks so that the app can be uploaded to Apple.

`# skip if we run in debug
if [ "$CONFIGURATION" == "Debug" ]; then
    echo "Skip frameworks cleaning in debug version"
    exit 0
fi

APP_PATH="${TARGET_BUILD_DIR}/${WRAPPER_NAME}"

# This script loops through the frameworks embedded in the application and
# removes unused architectures.
find "$APP_PATH" -name '*.framework' -type d | while read -r FRAMEWORK
do
FRAMEWORK_EXECUTABLE_NAME=$(defaults read "$FRAMEWORK/Info.plist" CFBundleExecutable)
FRAMEWORK_EXECUTABLE_PATH="$FRAMEWORK/$FRAMEWORK_EXECUTABLE_NAME"
echo "Executable is $FRAMEWORK_EXECUTABLE_PATH"

EXTRACTED_ARCHS=()

for ARCH in $ARCHS
do
echo "Extracting $ARCH from $FRAMEWORK_EXECUTABLE_NAME"
lipo -extract "$ARCH" "$FRAMEWORK_EXECUTABLE_PATH" -o "$FRAMEWORK_EXECUTABLE_PATH-$ARCH"
EXTRACTED_ARCHS+=("$FRAMEWORK_EXECUTABLE_PATH-$ARCH")
done

echo "Merging extracted architectures: ${ARCHS}"
lipo -o "$FRAMEWORK_EXECUTABLE_PATH-merged" -create "${EXTRACTED_ARCHS[@]}"
rm "${EXTRACTED_ARCHS[@]}"

echo "Replacing original executable with thinned version"
rm "$FRAMEWORK_EXECUTABLE_PATH"
mv "$FRAMEWORK_EXECUTABLE_PATH-merged" "$FRAMEWORK_EXECUTABLE_PATH"

done`

VitamineC, 2019-01-09:
@dantetwc I try used your fork as main library. I can build app to simulator however I still get code signing "Zip.framework" failed.

VitamineC, 2019-01-09:
@hokasan probably u get some progress? I can build archive, however when I try publish it to test fly I get zip.framework error

hokasan, 2019-01-09:
@VitamineC I can either build for simulator or create archive. The script @dantetwc pasted above is probably needed to do both....I just don't think it should be needed...

VitamineC, 2019-01-09:
@hokasan 
As for me - no. I сould create archive or build before @dantetwc code. As I understand his code should help up publish app to testfly

dantetwc, 2019-01-09:
To upload to AppStore or TestFlight, add the script as a run script
[image]

[image]

dantetwc, 2019-01-09:
> @VitamineC What I mean is that before @dantetwc PR I could build/run/create archive and after his patch I can create an archive that I can publish on appstore. The problem is that with the PR that enables me to publish to appstore I cannot build for the simulator. The proposed solution by @dantetwc (as I understand it) is to remove the simulator binarys when building Release version. This will probably work, but it seems like a hack to me and we haven't fully understood the problem.

This is a bit hacky but the only solution for now unless we link the iOSDFULibrary and Zip framworks to the XCode project. To be able to build for simulator, the frameworks have to be embedded with x86_64 and i386 arch (As the simulator runs on Mac platform), however, when we upload to AppStore, it will check for code that's not supposed to run on the iOS devices and hence it will say ode signing "Zip.framework" failed. just like @VitamineC experienced. Therefore, the only way is to merge two platform code into single framework file and remove it when necessary.

VitamineC, 2019-01-09:
@dantetwc thx you a lot. With your fork and run script I can build and publish it to test fly

hokasan, 2019-01-11:
@dantetwc Thanks for your help, I will try to use your workaround until a proper solution is available.

Looveh, 2019-04-16:
This should be resolved with v3.0.0 when using CocoaPods

closed 2019-04-16 by Looveh