Pilloxa/react-native-nordic-dfu #51: Error while start DFU. Error: DFU Service not found

Manish123Pandia, 2019-02-14:
Here is my code : 
   NordicDFU.startDFU({
                            deviceAddress: obj.MacAddress,
                            deviceName: obj.DeviceName,
                            filePath: filePath
                        }).then((res) => {
                            me.emit('showDfuUpgradeMessage');
                        }).catch((err) => {
                            BleUtil.log("Error while start DFU. " + err);
                            me.emit('hideDFULoader');
                            me.showServiceStatus('Not able to connect device');
                            return;

Dependices : 
 "react-native-nordic-dfu": "^1.2.1",
  "react-native-ble-plx": "^0.9.1",

i was gatting CBcentralMeneger issue, which I have fixed  
[image]

Manish123Pandia, 2019-02-14:
i am using iOS 12.1 version

Looveh, 2019-02-15:
@Manish123Pandia Unfortunately we don't have the time or resources to help debug your issue, though on the surface it seems to me that your app hasn't found a DFU BLE Service on the device you're connected to.

Manish123Pandia, 2019-02-22:
Looveh - Still the error is not resolving can you please help me now .

Manish123Pandia, 2019-02-27:
This is working fine in case of Android but having issue in the case of IOS

Manish123Pandia, 2019-02-27:
This is also working in IOS 10.3, this issue comes in latest version of application

Looveh, 2019-04-16:
I see that you're using an old version of this library, please upgrade to v3.0.0

closed 2019-04-16 by Looveh

vivekspraxa, 2019-07-24:
I am using the latest version 3.0.1 and still getting this issue

DFU state: CONNECTING
98d3db5e-35d2-4003-9…-fc749e7a9a6c:92442 DFU state: DFU_FAILED
98d3db5e-35d2-4003-9…5-fc749e7a9a6c:5221 Error: DFU Service not found
    at createErrorFromErrorData (98d3db5e-35d2-4003-9…5-fc749e7a9a6c:4204)
    at 98d3db5e-35d2-4003-9…5-fc749e7a9a6c:4156
    at MessageQueue.__invokeCallback (98d3db5e-35d2-4003-9…5-fc749e7a9a6c:4702)
    at 98d3db5e-35d2-4003-9…5-fc749e7a9a6c:4433
    at MessageQueue.__guard (98d3db5e-35d2-4003-9…5-fc749e7a9a6c:4606)
    at MessageQueue.invokeCallbackAndReturnFlushedQueue (98d3db5e-35d2-4003-9…5-fc749e7a9a6c:4432)
    at debuggerWorker.js:80