Pilloxa/react-native-nordic-dfu #56: Xcode 10.2 build fail – (dfu) binaries use incompatible versions of Swift

NN01010101, 2019-03-27:
After sneaky Xcode update to (10.2) it is no longer able to build the project complaining:

`error: The following binaries use incompatible versions of Swift:`
`.../node_modules/react-native-nordic-dfu/ios/iOSDFULibrary.framework/iOSDFULibrary`

I tried to clean the project and to remove DerivedData without success.

hokasan, 2019-04-03:
I have the same problem..

LiuMengzhou, 2019-04-12:
```
xxx/node_modules/react-native-nordic/react-native-nordic/Zip.framework/Zip compiled with older version of Swift language (unknown ABI version 0x06) than previous files (unknown ABI version 0x07) file 'xxx/node_modules/react-native-nordic/react-native-nordic/Zip.framework/Zip' for architecture x86_64
```
```
```

Looveh, 2019-04-16:
This should be resolved with v3.0.0

closed 2019-04-16 by Looveh

MD3XTER, 2019-04-24:
If you want to use the latest `ZIPFoundation.framework` and `iOSDFULibrary.framework` and also make it work on the new **Xcode 10.2.1**. I did a [fork](https://github.com/MD3XTER/react-native-nordic-dfu) of the `v2.0.1` version and have updated both frameworks to the latest `iOSDFULibrary - 4.4.2` & `ZIPFoundation - 0.9.9`. Because the CocoaPods method doesn't work for me and it messes up my entire project. This is the only way I found to have my project updated and not to break everything.