Pilloxa/react-native-nordic-dfu #57: Issue With Zip.framework: dyld: Library not loaded: @rpath/libswiftCore.dylib

philipyun, 2019-03-29:
Ran into an issue with the build process.  I followed all the instructions to manually link in iOS but the app crashes on start, being unable to load the libswiftCore.dylib lib file.

Please advise on how to resolve this issue.

t3hMarK, 2019-04-03:
I have the same issue since updating to the latest version of MacOS & Xcode. The build is successful and app crashes with a SIGABRT at launch.

Mac OS : 10.14.4
Xcode : 10.2
react-native : 0.59.3
react-native-ble-manager : 6.6.2
react-native-nordic-dfu : 2.0.1

`dyld: Library not loaded: @rpath/libswiftCore.dylib
  Referenced from: /private/var/containers/Bundle/Application/6DF0B37A-2656-4C08-B838-AE452CD3F77D/cleantest.app/Frameworks/Zip.framework/Zip
  Reason: image not found`

I get the same issue on a virgin project created via `react-native init [project name]`. I've added `react-native-ble-manager` & `react-native-nordic-dfu` via `yarn add` and followed step 4-10. Xcode Build Options have set `Always Embed Swift Standard Libraries` to yes.

Looveh, 2019-04-16:
This should be resolved with v3.0.0

closed 2019-04-16 by Looveh

MD3XTER, 2019-04-24:
If you want to use the latest `ZIPFoundation.framework` and `iOSDFULibrary.framework` and also make it work on the new **Xcode 10.4.1**. I did a [fork](https://github.com/MD3XTER/react-native-nordic-dfu) of the `v2.0.1` version and have updated both frameworks to the latest `iOSDFULibrary - 4.4.2` & `ZIPFoundation - 0.9.9`. Because the CocoaPods method doesn't work for me and it messes up my entire project. This is the only way I found to have my project updated and not to break everything.

alanwong0922, 2024-04-09:
> If you want to use the latest `ZIPFoundation.framework` and `iOSDFULibrary.framework` and also make it work on the new **Xcode 10.2.1**. I did a [fork](https://github.com/MD3XTER/react-native-nordic-dfu) of the `v2.0.1` version and have updated both frameworks to the latest `iOSDFULibrary - 4.4.2` & `ZIPFoundation - 0.9.9`. Because the CocoaPods method doesn't work for me and it messes up my entire project. This is the only way I found to have my project updated and not to break everything.

Hi andrei-zgirvaci , I have a same problem with my project. 
Can you tell me more about the solution of the react native with ZIPFoundation? 
On the other hands, The link https://github.com/MD3XTER/react-native-nordic-dfu would not found.