Pilloxa/react-native-nordic-dfu #59: iOS - dyld`__abort_with_payload: after link react-native-nordic-dfu and manually adding zip.framwork and iOSDFULibrary.Framework

pooja-chothe, 2019-04-12:
In my react-native project i'm using 'react-native-nordic-dfu' npm module. I have done react-native link react-native-nordic-dfu as well as manual installation steps(4-10). But after running the app it crash and throws Thread 1: signal SIGABRT dyld`__abort_with_payload: exception. I am not able to find any solution for that. Please let me know if anyone has solution for this.

Thanks in advance.

[image]

Looveh, 2019-04-16:
Hi, try using CocoaPods and v3.0.0 that was released yesterday, this is the new supported way of iOS integration

closed 2019-04-16 by Looveh

MD3XTER, 2019-04-24:
If you want to use the latest `ZIPFoundation.framework` and `iOSDFULibrary.framework` and also make it work on the new **Xcode 10.2.1**. I did a [fork](https://github.com/MD3XTER/react-native-nordic-dfu) of the `v2.0.1` version and have updated both frameworks to the latest `iOSDFULibrary - 4.4.2` & `ZIPFoundation - 0.9.9`. Because the CocoaPods method doesn't work for me and it messes up my entire project. This is the only way I found to have my project updated and not to break everything.

pooja-chothe, 2019-05-23:
I have used react-native-nordic-dfu@2.0.1 but still it throwing same error.

shayhz, 2019-09-03:
> I have used react-native-nordic-dfu@2.0.1 but still it throwing same error.
> It throws the error "This application’s bundle identifier does not match its code signing identifier xcode error because of adding framework". I think framework bundle identifier is not matched with app id.

I experience the same issue. Any solution for that ?