Pilloxa/react-native-nordic-dfu #62: Error on pod install

SebastianGrzymala, 2019-04-17:
Steps to reproduce:
```
react-native init AwesomeProject
cd AwesomeProject
npm install --save react-native-nordic-dfu
cd ios
touch Podfile
```
My Podfile:

```
target 'AwesomeProject' do
 rn_path = '../node_modules/react-native'

 # See http://facebook.github.io/react-native/docs/integration-with-existing-apps.html#configuring-cocoapods-dependencies
 pod 'yoga', path: "#{rn_path}/ReactCommon/yoga/yoga.podspec"
 pod 'React', path: rn_path, subspecs: [
   'Core',
   'CxxBridge',
   'DevSupport',
   'RCTActionSheet',
   'RCTAnimation',
   'RCTGeolocation',
   'RCTImage',
   'RCTLinkingIOS',
   'RCTNetwork',
   'RCTSettings',
   'RCTText',
   'RCTVibration',
   'RCTWebSocket',
 ]

 # React Native third party dependencies podspecs
 pod 'DoubleConversion', :podspec => "#{rn_path}/third-party-podspecs/DoubleConversion.podspec"
 pod 'glog', :podspec => "#{rn_path}/third-party-podspecs/glog.podspec"
 # If you are using React Native <0.54, you will get the following error:
 # "The name of the given podspec `GLog` doesn't match the expected one `glog`"
 # Use the following line instead:
 #pod 'GLog', :podspec => "#{rn_path}/third-party-podspecs/GLog.podspec"
 pod 'Folly', :podspec => "#{rn_path}/third-party-podspecs/Folly.podspec"

 pod 'react-native-nordic-dfu', :path => '../node_modules/react-native-nordic-dfu'

end

post_install do |installer|
 installer.pods_project.targets.each do |target|
   if target.name == "React"
     target.remove_from_project
   end
 end
end

```

Install pods:
```
pod install
```

I get an error:

```
[!] Unable to determine Swift version for the following pods:
 - `iOSDFULibrary` does not specify a Swift version and none of the targets (`AwesomeProject`) integrating it have the `SWIFT_VERSION` attribute set. Please contact the author or set the `SWIFT_VERSION` attribute in at least one of the targets that integrate this pod.
```

SebastianGrzymala, 2019-04-17:
I can set SWIFT_VERSION in xCode build settings. It resolves pod install problem, but then during the build I get an error:

`fatal error: module map file '/Users/user/Library/Developer/Xcode/DerivedData/AwesomeProject-fotojjsqmaekwtaxiqwdgagnlkci/Build/Products/Debug-iphonesimulator/ZIPFoundation/ZIPFoundation.modulemap' not found`

Looveh, 2019-04-17:
This is a common CocoaPods issue, try completely cleaning your project, including deleting the `Pods` directory

SebastianGrzymala, 2019-04-17:
@Looveh Thank you for your answer. I have created a new project for the purpose of this issue. There were no any conflict cache files there. I removed the `Pods`, `podfile.lock` and `build` folder anyway  but it still hasn't worked.

wimva, 2019-04-18:
+1 on this. Got a lot of trouble updating the package for Xcode 10.2. Also tried to setup a new project, but still can't get it to work.

dariuszseweryn, 2019-04-19:
I have successfully updated the project from `2.x.x` to `3.0.0` but it took me significant time to do it properly.

It would be awesome to have a migration guide prepared. I needed to search for [the commit that changed README to the new version](https://github.com/Pilloxa/react-native-nordic-dfu/commit/e7d8fcaf0eb5858f2181456508e0720f453d5113#diff-04c6e90faac2675aa89e2176d2eec7d8) and reverse the steps.

Additionally when doing bigger changes in a project I try to make a _totally clean install_ i.e.:
- remove `android/.gradle` caches
- remove `ios/Pods` files
- remove `node_modules`
- remove `$TMPDIR/react-*`
- remove `~/Library/Caches/Cocoapods`
- remove `~/Library/Developer/Xcode/DerivedData`
And only then `npm install`

If logs would say that `pods` did fail at some step they also say what to do e.g.
```
You have either:
 * out-of-date source repos which you can update with `pod repo update` or with `pod install --repo-update`.
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.

Note: as of CocoaPods 1.0, `pod repo update` does not happen on `pod install` by default.
```
One has to follow the steps.

Hope this helps

stonehippo, 2019-04-19:
@dariuszseweryn Did you encounter issues with `Module 'iOSDFULibrary' not found`? If so, how did you get past that. I've pretty much cleaned everything that can be, but the build keeps complaining that the include of `@import iOSDFULibrary;` in `RNNordicDfu.h` won't resolve.

dariuszseweryn, 2019-04-19:
> Did you encounter issues with `Module 'iOSDFULibrary' not found`?

I do not recall this issue. Have you tried to clean the XCode project? `Cmd+Shift+k` and `Cmd+Alt+Shift+k`?

stonehippo, 2019-04-19:
Yeah, I've been working with Xcode for a decade+; cleaning was the first thing I did. But thanks for the suggestion. ;-) 

It's something to do with the linked libraries, I think.

dariuszseweryn, 2019-04-20:
Hmmm. So maybe `react-native link react-native-nordic-dfu` again?

SchwarzerSchnee, 2019-04-23:
When I try to use this lib in a fresh RN project, I get the `Module 'iOSDFULibrary' not found` that others here are getting too. And when I try to use this new version of the lib in my real project that has used the old one before, I get `'React/RCTBridgeModule.h' file not found` instead. For both of them I have no idea how to fix them. The latter has plenty of results on Internet search, but none of what's suggested there works. It looks like this update that was meant to make things easier has made them more complicated instead. How do we do our firmware updates now and/or when do we get them back to working?

Looveh, 2019-04-24:
@SchwarzerSchnee XCode can be a minefield to navigate since everybody's project setup is pretty much unique to their project. Is the issue resolved if you in XCode > Your Target > Build Settings add a User-Defined build setting `SWIFT_VERSION = 5.0`?

Looveh, 2019-04-24:
@dariuszseweryn Thanks, I've updated the release notes with your instructions, please let me know if there's another step that solved any issues for you and I'll add them too :+1:

SchwarzerSchnee, 2019-04-25:
@Looveh Doesn't work, changes nothing on both the real project and the fresh project.

Looveh, 2019-04-25:
@SchwarzerSchnee I've updated the release notes with update instructions, does following these steps resolve your issue? https://github.com/Pilloxa/react-native-nordic-dfu/releases/tag/v3.0.0

SchwarzerSchnee, 2019-04-25:
Already seen and done that, and no, also didn't help.

SebastianGrzymala, 2019-04-25:
I tried to run a project from an example folder. Android & iOS failed.

**Android Studio 3.3:**
```
ERROR: Gradle DSL method not found: 'google()'
Possible causes:
The project 'android' may be using a version of the Android Gradle plug-in that does not contain the method (e.g. 'testCompile' was added in 1.1.0).
Upgrade plugin to version 3.3.2 and sync project

The project 'android' may be using a version of Gradle that does not contain the method.
Open Gradle wrapper file

The build file may be missing a Gradle plugin.
Apply Gradle plugin
```

**Xcode 9.4.1:**

Pods installed successfully.
The SWIFT VERSION is set to 5 by default in `Build Settings` -> `User definied`
Get an error:
```
The “Swift Language Version” (SWIFT_VERSION) build setting must be set to a supported value for targets which use Swift. This setting can be set in the build settings editor.
```

Looveh, 2019-04-25:
@SchwarzerSchnee @SebastianGrzymala Have you added `use_frameworks!` to your Podfile?

SebastianGrzymala, 2019-04-25:
In my case adding `use_frameworks!` caused other error:
```
The 'Pods-MyAppName' target has transitive dependencies that include static binaries: (/Users/user/Desktop/******/******/ios/Pods/GoogleMaps/Base/Frameworks/GoogleMapsBase.framework
```
I found a fix which seems to be working:
```
# podfile
pre_install do |installer|
    # workaround for https://github.com/CocoaPods/CocoaPods/issues/3289
    Pod::Installer::Xcode::TargetValidator.send(:define_method, :verify_no_static_framework_transitive_dependencies) {}
end
```
Pods install runs successfully, but I have a problem with build now. 
I looks that `use frameworks!` breaks some popular libraries:
In my case I get an error 
```
'FirebaseCore/FirebaseCore.h' file not found
```
The problem with `use framework!` is described [here](https://github.com/invertase/react-native-firebase/issues/1166#issuecomment-394162920).

Looveh, 2019-04-26:
@SebastianGrzymala I see, yes this is a common problem when using CocoaPods and Swift libraries. Unfortunately this is an issue caused by the architecture of the library `iOSDFULibrary` which this library depends on.

SebastianGrzymala, 2019-04-28:
https://github.com/Pilloxa/react-native-nordic-dfu/commit/86c2194c0a49324edf0810a51eac3ff21be80e8e

closed 2019-04-28 by SebastianGrzymala