Pilloxa/react-native-nordic-dfu #63: Error on Build version 3.0.0

philipyun, 2019-04-19:
Having an issue building with this library using the 3.0.0

Followed all instructions in the readme

```
yarn add react-native-nordic-dfu
react-native link react-native-nordic-dfu
```

Created my podfile with pod init, added the path to the nordic dfu library.

Build with Xcode version 10.2.

Xcode Complains about RNNordicDFU.h not found. So I added the react-native-nordic-dfu to my search path.  Now it complains that iOSDFULibrary is not found.

The readme is clearly missing information on how to successfully build a project with this repo

Manish123Pandia, 2019-04-22:
Facing the same issue

MD3XTER, 2019-04-24:
If you want to use the latest `ZIPFoundation.framework` and `iOSDFULibrary.framework` and also make it work on the new **Xcode 10.2.1**. I did a [fork](https://github.com/MD3XTER/react-native-nordic-dfu) of the `v2.0.1` version and have updated both frameworks to the latest `iOSDFULibrary - 4.4.2` & `ZIPFoundation - 0.9.9`. Because the CocoaPods method doesn't work for me and it messes up my entire project. This is the only way I found to have my project updated and not to break everything.

Looveh, 2019-04-24:
@philipyun Seems like you haven't defined which Swift version to compile your project with, try adding it under Build Settings > User Defined settings (ex. `SWIFT_VERSION = 5.0`) and see if that works

Looveh, 2019-04-24:
@MD3XTER Be warned that ad-hoc compiled framework files has broken with pretty much every XCode upgrade in the past and will likely continue to do so for every major XCode release in the future as well. We understand that CocoaPods might not be an ideal setup for everybody, however, finding a stable solution to this issue is beyond my understanding of the XCode toolchain, so any stable solution with pre-built framework files is welcome!

shengliZh, 2019-04-25:
@MD3XTER @Looveh 
Have you encountered this problem when running directly on an iPhone?
"App installation failed"
"This application’s bundle identifier does not match its code signing identifier."
[image]

MD3XTER, 2019-04-25:
@shengliZh I don't think this has anything to do with the library.

SchwarzerSchnee, 2019-04-25:
@shengliZh This has helped me. https://stackoverflow.com/a/52008848

@MD3XTER I got your fork to build and run, but when I actually try to do the device update, the device just disconnects and the update never starts, when it obviously has worked before. Any ideas?

MD3XTER, 2019-04-25:
@SchwarzerSchnee hmm, I didn't actually test the DFU process yet. I just needed it to build and run. I would try to search maybe there is some issue with the latest update of the [iOSDFULibrary.framework](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues?utf8=✓&q=). Maybe try to downgrade to a lower version and see if that helps?

shengliZh, 2019-04-26:
@SchwarzerSchnee I have test the fork of @MD3XTER  on ios and android, both works well.

MD3XTER, 2019-04-29:
@shengliZh I actually did try to build for my phone and it gave me the same error that you have with `This application's bundle identifier does not match its code signing identifier.` So I think it has to do something with this library, or we are unlucky. It does happen only when I build for the device. For simulator it works just fine.

MD3XTER, 2019-04-29:
I managed to fix this by doing the [following](https://github.com/Pilloxa/react-native-nordic-dfu/issues/63#issuecomment-486655918)

wimva, 2019-05-02:
I'm getting `'React/RCTBridgeModule.h' file not found` in RNNordicDfu.h
Anyone any idea to solve this one?

Looveh, 2019-05-02:
@wimva Have you removed the two `.framework` files completely from within XCode?

wimva, 2019-05-02:
@Looveh Yes, these are completely gone already

Looveh, 2019-05-02:
`'React/RCTBridgeModule.h' file not found` suggests that either CocoaPods wasn't able to properly set the search paths for your project, which would happen if you've manually modified search paths from within XCode and not added `$(inherited)`. `pod install` would give you these warnings though

wimva, 2019-05-02:
`$(inherited)` is in there, also no errors on `pod install`. I'll wipe everything and try again.

wimva, 2019-05-03:
@Looveh Whatever I do, I still get the same issue `'React/RCTBridgeModule.h' file not found`. The only warning I get after `pod install` is `[!] React has been deprecated`

SchwarzerSchnee, 2019-05-03:
Unfortunately, I did not fix this at all and have lost more than a week over this by now. I'm facing the same issues as you, without any clues on how to deal with them.

Looveh, 2019-05-07:
If you're able to find the issue, please submit a PR with the fix or documentation on how to avoid the issue