Pilloxa/react-native-nordic-dfu #64: ios build error

shengliZh, 2019-04-24:
[image]

MD3XTER, 2019-04-24:
If you want to use the latest `ZIPFoundation.framework` and `iOSDFULibrary.framework` and also make it work on the new **Xcode 10.4.1**. I did a [fork](https://github.com/MD3XTER/react-native-nordic-dfu) of the `v2.0.1` version and have updated both frameworks to the latest `iOSDFULibrary - 4.4.2` & `ZIPFoundation - 0.9.9`. Because the CocoaPods method doesn't work for me and it messes up my entire project. This is the only way I found to have my project updated and not to break everything.

Looveh, 2019-04-24:
@shengliZh This is likely due to an old bug in React Native's CocoaPods setup that has been described here https://github.com/facebook/react-native/issues/23390 and shouldn't be caused by this library

shengliZh, 2019-04-25:
@Looveh Thank you so much, it looks ok.

shengliZh, 2019-04-25:
@MD3XTER Thank you so much, your branch works well also.

closed 2019-04-25 by shengliZh