Pilloxa/react-native-nordic-dfu #66: Build Error on Version 3.0.0

hanheuljo, 2019-04-25:
I followed the instructions on `README.md`, down to adding `SWIFT_VERSION = 5.0`, but it says 
```
Module 'iOSDFULibrary' not found
```

I'm currently using `react-native version 0.58.6`

Looveh, 2019-04-26:
@hanheuljo Did you follow the upgrade instructions https://github.com/Pilloxa/react-native-nordic-dfu/releases/tag/v3.0.0 and add `use_frameworks!` to your Podfile?

hanheuljo, 2019-04-26:
The Podfile I'm using is directly from your example app, and I've tried to add the package in a new projet, which still did not work

Looveh, 2019-04-29:
Are you having this issue when loading the JS code in your app, when compiling natively on iOS or natively when the app just started on iOS?

Looveh, 2019-04-29:
I'm able to get the example project running on my machine, are you sure you've cleaned your project and removed all non-Pod related references to the library?

SebastianGrzymala, 2019-04-30:
@Looveh I managed to run the example project only when update the Xcode. SWIFT version 5 was implemented in 10.2 which is very new. I think you should update readme to use minimum this version.

Looveh, 2019-05-02:
@SebastianGrzymala Thanks, I've updated the README with this in mind

SchwarzerSchnee, 2019-05-03:
I'm still having the same issue. Example project builds and runs fine, but adding RNNDFU to a fresh react-native-init project (or to a bigger and real project) does not work, errors alternating between "iOSDFULibrary not found", "RCTBridgeModule.h not found", and 52 undefined symbols of the kind _JS*. I have compared various project settings to each other and made sure they would be the same as in the example, but even then, it simply will not work and I can't wrap my head around it.

Please, we need to get this resolved asap, can't update our app on the store without this working.

Looveh, 2019-05-03:
@SchwarzerSchnee Are you able to run the example project?

SchwarzerSchnee, 2019-05-03:
@Looveh 
>Example project builds and runs fine
*Only* the example project.

Looveh, 2019-05-03:
@SchwarzerSchnee I'm afraid I won't be able to help you any further then if you've cross checked your build settings between the example project and your project, there must be some difference in settings that's causing your issue.

NN01010101, 2019-06-23:
@Looveh 
Hi, I have the exact same problem and I found a workaround. 
I started a fresh project and installed the react-native-nordic-dfu library as described in readme, no luck, project wouldn't compile. 
So I cloned the the react-native-nordic-dfu repo outside of my project and changed the path in the Podfile of my project pointing to the repo like so ../../react-native-nordic-dfu. Repeat _pod install_ and now my xcode build succeeded. 
I'm not sure what is missing in the npm project, but there must be something different. This explains why only the example project compiles.

Looveh, 2019-06-24:
@NN01010101 Very interesting, can you see which files diff between the two installations? And what build error do you get? We have a project that does compile when installed through npm so it's not only the example project that compiles.

nvelozsavino, 2019-06-28:
I had the same issue where I was getting the error: Module 'iOSDFULibrary' not found
After a lot of tests I found that the problem might be the order of commands. 
If there's no Podfile before executing the react-native link command, the RNNordicDFU.xcproject will be added to the Libraries folder on the XCode project, and this is what's causing the iOSDFULibrary not found issue, it's like the project will compile twice the react-native-nordic-dfu library, once with the RNNordicDFU.xcproject and again with the Pods.

If you create the Podfile with only the react-native-nordic-dfu pod before executing the react-native link command, the RNNordicDFU.xcproject will not be added to the libraries folder on the XCode but it will then fail in the include RCTBridgeModule.h because the pod project seems to not have the search path, this might be fixed adding something to the Header Search Path, however I fixed adding the React pod in the podfile like in the example like this:

```
platform :ios, "9.0"

# External dependencies contain Swift, important to add this line
use_frameworks!

target "NordicDFUExample" do
  # Standard React Native dependencies when using CocoaPods
  pod "yoga", path: "../node_modules/react-native/ReactCommon/yoga"
  pod "DoubleConversion", podspec: "../node_modules/react-native/third-party-podspecs/DoubleConversion.podspec"
  pod "glog", podspec: "../node_modules/react-native/third-party-podspecs/glog.podspec"
  pod "Folly", podspec: "../node_modules/react-native/third-party-podspecs/Folly.podspec"
  pod "React", path: "../node_modules/react-native", subspecs: [
        "Core",
        "CxxBridge",
        "DevSupport",
        "RCTBlob",
        "RCTImage",
        "RCTLinkingIOS",
        "RCTText",
      ]

  pod "react-native-ble-manager", path: "../node_modules/react-native-ble-manager"

  # replace this path with "../node_modules/react-native-nordic-dfu" or wherever
  # it's installed in your project
  pod "react-native-nordic-dfu", path: "../node_modules/react-native-nordic-dfu"
end
```
Something similar happened to me with the react-native-ble-manager that I added to the Podfiles first, then the BleManager.xcproject was not added to the Libraries, however I updated the Podfile to comment it, then add it manually, but it failed because the header search paths wern't correct and it couldn't find the BleManager.h called from the AppDelegate.m, so I remove the BleManager.xcproject and add the react-native-ble-manager pod to the Podfile and it build succesfully.

There's one thing I need to try because I managed to build successfully before removing the RNNordicDFU.xcproject manually but I ended up with a Native module not found error when running the application and I'm not sure it was a problem with that.

nvelozsavino, 2019-06-28:
Even though I manage to build the project, it is not correctly linked in react native, every time I try to import DFUEmitter, DFUStatus, NordicDFU in javascript file, I get the "Native module cannot be null" on runtime. This as far as I seen is caused when a native library is not linked properly

nvelozsavino, 2019-06-30:
I manage to build and run the project successfully. These are the steps I followed:

1. Create the ios/Podfile before executing react-native link
1. Add react-native-nordic-dfu and all other libraries you need to the Podfile
1. Add to the Podfile all the React items located on the Libraries folder of the XCode project
```
platform :ios, "9.0"

# External dependencies contain Swift, important to add this line
use_frameworks!

target "project" do

  pod 'React', :path => '../node_modules/react-native', :subspecs => [
    'Core',
    'CxxBridge',
    'DevSupport',
    # the following ones are the ones taken from "Libraries" in Xcode:
    'RCTAnimation',
    'RCTActionSheet',
    'RCTBlob',
    'RCTGeolocation',
    'RCTImage',
    'RCTLinkingIOS',
    'RCTNetwork',
    'RCTSettings',
    'RCTText',
    'RCTVibration',
    'RCTWebSocket'
  ]

  # Standard React Native dependencies when using CocoaPods
  pod "yoga", path: "../node_modules/react-native/ReactCommon/yoga"
  pod "DoubleConversion", podspec: "../node_modules/react-native/third-party-podspecs/DoubleConversion.podspec"
  pod "glog", podspec: "../node_modules/react-native/third-party-podspecs/glog.podspec"
  pod "Folly", podspec: "../node_modules/react-native/third-party-podspecs/Folly.podspec"

  pod "react-native-ble-manager", path: "../node_modules/react-native-ble-manager"
  pod "react-native-fetch-blob", path: "../node_modules/react-native-fetch-blob"

  # replace this path with "../node_modules/react-native-nordic-dfu" or wherever
  # it's installed in your project
  pod "react-native-nordic-dfu", path: "../node_modules/react-native-nordic-dfu"

  # Add here other libraries
end
```
1. Execute react-native link
1. Execute pod install
1. Open the project xcworkspace created by the pod install
1. Remove all the items located on the Libraries folder in the XCode project (remove references). (I don't know what would happen if there's a librarie that can not be add to the Podfile, I suppose that you shouldn't delete it from the Libraries folder, however it might fail looking for references)
1. Set Always Embed Swift Standard Libraries on the target project in Build Settings (I'm not sure this is required but in older versions this was a step required to use the iOSDFULibrary) 
1. Set SWIFT_VERSION to the required value for your XCode, I added this to the end of the Podfile because I needed the version 4.2:
```
post_install do |installer|
  installer.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
      config.build_settings['SWIFT_VERSION'] = '4.2'
    end
  end
end
```
1. Build and execute