Pilloxa/react-native-nordic-dfu #67: Unhandled JS Exception: null is not an object (evaluating 'RNNordicDfu.startDFU')

shengliZh, 2019-05-16:
This error occured on ios with react-native-nordic-dfu 3.0.0

Looveh, 2019-05-16:
Can you share your code to provide some more context?

shengliZh, 2019-05-16:
[image]
Is there any error? But it works well on Android

shengliZh, 2019-05-16:
[image]
This is the log.

Looveh, 2019-05-16:
It crashes on this line https://github.com/Pilloxa/react-native-nordic-dfu/blob/master/index.js#L46 which it will do if the library hasn't been linked properly

shengliZh, 2019-05-17:
[image]
function startDFU({ deviceAddress, deviceName, filePath }) {
  if (deviceAddress == undefined) {
    return rejectPromise('No deviceAddress defined');
  }
  if (deviceName == undefined) {
    return rejectPromise('No deviceName defined');
  }
  if (filePath == undefined) {
    return rejectPromise('No filePath defined');
  }
  const upperDeviceAddress = deviceAddress.toUpperCase();
  console.log('RNNordicDfu result: ' + RNNordicDfu);
  return RNNordicDfu.startDFU(upperDeviceAddress, deviceName, filePath);
}

Why the 'RNNordicDfu' is null?