Pilloxa/react-native-nordic-dfu #69: Error DFU FILE NOT FOUND

ArunaJithesh, 2019-06-01:
While doing 
NordicDFU.startDFU({
      deviceAddress: DEVICE_ID,
      name: "Ep",
      filePath: '/data/user/0/com.nordicdfuexample/files/rose.jpg'
    })
Causing error "DFU FILE NOT FOUND"
But file is there in file path.Please help me sort out this issue

JohnathanHill, 2019-06-14:
I had this same issue but fixed it using [react-native-fs](https://www.npmjs.com/package/react-native-fs/v/1.2.0). You should try using RNFS.copyFileAssets to copy the file to the device, then use RNFS.existsAssets to see if the file is pulling from your projects assets folder. Finally, use RNFS.exists to verify that it was copied to the device and RNFS.readDir to get a list of the files in the Device and see if it's there. I was able to see that I was copying the file to the device incorrectly using these. Hope this helps.

JohnathanHill, 2019-06-14:
> While doing
> NordicDFU.startDFU({
> deviceAddress: DEVICE_ID,
> name: "Ep",
> filePath: '/data/user/0/com.nordicdfuexample/files/rose.jpg'
> })
> Causing error "DFU FILE NOT FOUND"
> But file is there in file path.Please help me sort out this issue

I had this same issue but fixed it using react-native-fs. You should try using RNFS.copyFileAssets to copy the file to the device, then use RNFS.existsAssets to see if the file is pulling from your projects assets folder. Finally, use RNFS.exists to verify that it was copied to the device and RNFS.readDir to get a list of the files in the Device and see if it's there. I was able to see that I was copying the file to the device incorrectly using these. Hope this helps.

blakeparkinson, 2019-10-28:
@JohnathanHill Do you remember what you did here? I'm able to render in image from the exact same path in the UI, but if I call `startDFU` - I get the same "DFU FILE NOT FOUND" error. I've been using the file system and can clearly see that the file exists in the path.

Any guidance or help would be appreciated, we've been stuck on this for a couple days now.

rodrigosimoesrosa, 2019-11-08:
Hi everyone,
I've lost a lot of time with this, but I found a solution that worked for me.

You only need to add this line in proguard-rules.pro file:

-keep class no.nordicsemi.android.dfu.** { *; }

Reference:
NordicSemiconductor/Android-DFU-Library#37

SaltedBlowfish, 2021-02-11:
If anyone else stumbles across the `DFU FILE NOT FOUND` error only happening on either Android or iOS but the proguard fix doesn't work for them, here was my issue:

Using the `file:///path` format only works on iOS for DFU package `filePath` parameter. On Android, I found that you have to format it like: `/path` (without the `file://` prefix).

Hope this helps someone else.

s10mcow, 2023-11-17:
@SaltedBlowfish YES! You saved me!