Pilloxa/react-native-nordic-dfu #72: Error on StartDFU iOS (only): DFU Service Not Found

JohnathanHill, 2019-06-25:
Project builds and runs perfectly on both platforms, however, when I start the DFU process on iOS, the error I get is "DFU Service Not Found". Any ideas as to what this error would pertain to? Any suggestions would be greatly appreciated. I am running the newest release of this library and it fully completes the DFU process on Android. Thanks in advance!

JohnathanHill, 2019-06-25:
Nevermind, I solved it.

closed 2019-06-25 by JohnathanHill

Looveh, 2019-06-26:
@JohnathanHill What was your issue and solution? Maybe it can help others here too :)

vivekspraxa, 2019-07-24:
> Nevermind, I solved it.

Do you have any solution , i am also getting same issue

JohnathanHill, 2019-07-24:
@vivekspraxa In my case, I was updating a Buttonless Device. I had to go into the actual iOSDFULibrary Pods in XCode and open DFUServiceInitiator.swift and set the flag "enableUnsafeExperimentalButtonlessServiceInSecureDfu" to true. Then, I set my workspace settings to "legacy build system" and did a rebuild. That seemed to work in my case. May not be a solution for you, but this fixed my issue.

RRMoelker, 2020-08-19:
Changing the `enableUnsafeExperimentalButtonlessServiceInSecureDfu` helped for me.

Let me reorganize your suggestion to be a bit more readable. Change the line:

```
@objc public var enableUnsafeExperimentalButtonlessServiceInSecureDfu = true
```

in the file:

xcode project > Pods > Pods > iOSDFULibrary > DFUServiceInitiator.swift