Pilloxa/react-native-nordic-dfu #74: startDFU fails with DFUFILE NOT FOUND on Android

stonehippo, 2019-07-25:
I'm trying to load a DFU file from a .zip file bundled within the app. The file sits at `[project]/android/app/src/main/assets/firmware/filename.zip`. On iOS, I'm using `react-native-fs` to get the path to the resource, and it's working as expected, but on Android, I can't figure out what path it might accept. Here's how it's working right now on iOS:

```
    const { localName, id } = deviceToUpdate
    const rootPath = Platform.OS == "ios" ? RNFS.MainBundlePath : "firmware"
    const fileName = 'firmware.zip'
    const filePath = `${rootPath}/${fileName}`
    const dfuConfig = { deviceAddress: id, name: localName, filePath }

    NordicDFU.startDFU(dfuConfig)
```

I can't figure out what path might work, and the docs aren't get about how the DFU path is getting resolved. Any one have an idea of how I can fix this?

stonehippo, 2019-07-25:
Nevermind, I ended up using `react-native-fs` to copy the files to another location.

hokasan, 2019-10-03:
@stonehippo I have the same issue. I just to work for me but suddenly stopped. I use this path "/data/user/0/com.radinn.debug/files/fw/thalamus/2.bin". To where did you copy the files for it to work?

blakeparkinson, 2019-10-29:
@hokasan were you able to have any luck? Running into the same issue here

rodrigosimoesrosa, 2019-11-08:
Hi everyone, 
I've lost a lot of time with this, but I found a solution that worked for me.

You only need to add this line in proguard-rules.pro file:

`-keep class no.nordicsemi.android.dfu.** { *; }`

Reference:
https://github.com/NordicSemiconductor/Android-DFU-Library/issues/37

SaltedBlowfish, 2021-02-11:
@rodrigosimoesrosa I've tried this solution multiple times to no avail. I can also confirm that the file exists on my end using the filesystem.

Did you have to do anything specific to clear Android build cache or anything after adding that line?