Pilloxa/react-native-nordic-dfu #78: iOS 13 breaking changes

NN01010101, 2019-10-12:
Looks like in iOS 13 Apple made some changes to the CBCentralManager. 
The issue happens at line 209 inside of RNNordicDfu.m
**retrievePeripheralsWithIdentifiers** returns 0 so the DFU process quits with error "_Could not find device with deviceAddress_"

As pointed out [here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/issues/322#issuecomment-517438862)  it's now mandatory to wait before the state CBCentralManager becomes poweredOn before calling theretrievePeripheralsWithIdentifiers

As a workaround, I put ` [NSThread sleepForTimeInterval: 2];` right after `CBCentralManager * centralManager = getCentralManager();` and it works.

Looveh, 2019-10-14:
@NN01010101 Great! If this is a problem that many others might face, feel free to create a PR with your fixes. If you could get it to work with something more general than a sleep for an arbitrary number of seconds I think it would be better suited for other's projects and setups as well.

NN01010101, 2019-10-14:
@Looveh the issue is definitely there for iOS 13. Unfortunately, I'm not that versatile in Objective C, so I won't be able to write anything better than a delay, but it looks like guys could [fix it here](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/pull/327/commits/a64315b516f5c9c8bb52451b4b277e79a1ec9640)

LackoDan, 2019-10-17:
We're also having this issue. Unfortunately, the delay does not help in our case -- it blocks the app.
I tried forcing the library to use iOSDFULibrary 4.5.1 (in which the [Bugfix for iOS 13](https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/pull/327/commits/a64315b516f5c9c8bb52451b4b277e79a1ec9640) was merged), but that did not change anything.

ThomasStubbe, 2019-10-17:
Same here. The sleep does makes the error to go away, but my app also just hangs...

Looveh, 2019-10-17:
For what it's worth I know we're successfully using the version 3.0.1 in our app without having to make any alterations for iOS 13. Not sure about what makes our setup successful.

ThomasStubbe, 2019-10-17:
I've managed to work around the problem, by introducing two sleeps + updating the iOS-DFU-Library to 4.5.1... I've forked this lib: https://github.com/ThomasStubbe/react-native-nordic-dfu

Jurpp, 2020-08-17:
@ThomasStubbe We've been using your fork to work around the iOS13 issue. Android still works fine like it used to, but on iOS the first try always fails on `[Error: X bytes were sent while X bytes were reported as received]`. When trying a second time it always works. Do you have similar experiences?

RRMoelker, 2020-10-20:
We've been running into the same iOS13 issue `[Error: X bytes were sent while Y bytes were reported as received]`.

The fork from @Jurpp did not mesh well other pods (some versions mismatch).

I created a recent fork with the same changes: https://github.com/DreamOnward/react-native-nordic-dfu

This fork solved the bytes sent issue for us.