Pilloxa/react-native-nordic-dfu #93: Error: Operation failed, DFU State: DFU_FAILED, nRF52832, Adafruit Bootloader

SaltedBlowfish, 2020-03-30:
Hello!

Looking for some guidance here. I have the library installed and "functional" but I believe I have identified an issue with the packets per connection interval causing the update to crash.

Error (from exception on phone):

```
Error: Operation failed
```

DFU logging in console:

```
LOG  DFU State: CONNECTING
LOG  DFU State: DFU_PROCESS_STARTING
LOG  DFU State: ENABLING_DFU_MODE
LOG  DFU State: DFU_PROCESS_STARTING
LOG  DFU State: DFU_STATE_UPLOADING
LOG  DFU progress: 0%
LOG  DFU State: DFU_FAILED
```

The Nordic nRF Connect app throws the same error but provides an error code, which I've come to realize happens when the phone sends too many packets. I turned the "PRN" option in the Nordic app down to 6 (from 12) and the firmware update works fine.

Seems like we need a way to throttle the speed of the upload for devices that can't handle it.

Here is some more info on my setup:

```
iOS 13.3.1
iPhone 11 Pro Max
nRF52832
Adafruit Bootloader: feather_nrf52832_bootloader-0.3.2_s132_6.1.1.hex (latest)
react-native-ble-manager: 7.2.0
react-native-nordic-dfu: 3.0.1
```

Any advice is appreciated! Thank you.

Looveh, 2020-04-01:
@SaltedBlowfish Great catch! I'd have a look at the underlying libraries provided by Nordic and see if there's support for throttling there:

https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library
https://github.com/NordicSemiconductor/Android-DFU-Library

SaltedBlowfish, 2020-04-01:
There does seem to be support for the exact same `PRN` value that they use in the `nRF Connect` app in both libraries.

iOS Example for Legacy DFU:
- https://github.com/NordicSemiconductor/IOS-Pods-DFU-Library/blob/d8d32fe5c4ccf229ae5d21c1055780ff259f20f0/iOSDFULibrary/Classes/Implementation/LegacyDFU/Peripherals/LegacyDFUPeripheral.swift

I'm going to see if I can resolve this before circling back to this issue again (but please keep open as I think it's a valid feature request).

**As a note to anyone using Adafruit's `nRF` bootloader that stumbles across this problem:**

The _real_ issue seems to be in Adafruit's bootloader for the `nRF`, and there are some GitHub issues and a PR for it here:

- https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/63
- https://github.com/adafruit/Adafruit_nRF52_Bootloader/pull/80

Still, would be awesome if this library supported the PRN system like the `nRF Connect`  app does.

p4bl1t0, 2022-04-25:
hey @SaltedBlowfish do you find a way to get it work? Any fix I could do right now instead of waiting to this to be implemented?

giobauermeister, 2022-04-27:
Hey @p4bl1t0 our app developer found the Package Receipt Notification value inside Nordic's iOS DFU Library. It's hardcoded as 12. It's too fast for Nordic chip to process de update.

Please have a look at:

https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/42b13e8b84404fd38eb50ae75c46559032706ce6/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L115

Our app developer changed that to 1 and now it's working good. He did a patch to change that file, changing the value to 1.

SaltedBlowfish, 2022-04-27:
Hello @p4bl1t0. I'm sorry for the delay.

I was able to get it to work by building a custom secure boot loader (from the Adafruit one I was using). If you don't have that option, changing that value to 6 should be sufficient. I'm glad @giobauermeister mentioned that their developer changed the value to 1 but that's overkill. I tested thoroughly and like I mentioned in my original post, 6 was working consistently for me.

I deployed my product and had no issues with OTA updates with this `react-native-nordic-dfu` library since. Many thanks to @Looveh  et. al. for the amazing work here.

## If you're reading this looking for a quick resolution

Read [the comment from @giobauermeister](https://github.com/Pilloxa/react-native-nordic-dfu/issues/93#issuecomment-1111133929) about the specific place to change the `12` to a `6` [here in the NordicSemiconductor library file](https://github.com/NordicSemiconductor/IOS-DFU-Library/blob/42b13e8b84404fd38eb50ae75c46559032706ce6/iOSDFULibrary/Classes/Implementation/DFUServiceInitiator.swift#L115). You'll want to make it a patch using [patch-package](https://www.npmjs.com/package/patch-package) until this library supports changing the setting dynamically from JavaScript.

closed 2022-04-27 by SaltedBlowfish

p4bl1t0, 2022-04-29:
Well getting the same error trying to doing DFU on Android 10 (moto g 7)

>  LOG  Starting DFU /data/user/0/com.evolutiondevices.ewalk/files/RNFetchBlobTmp_ig0uq3zpynnwpiqscg237.zip
>  LOG  DFU state: DFU_PROCESS_STARTING
>  LOG  DFU state: DFU_PROCESS_STARTING
>  LOG  DFU state: ENABLING_DFU_MODE
>  LOG  DFU state: ENABLING_DFU_MODE
>  LOG  DFU state: CONNECTING
>  LOG  DFU state: CONNECTING
>  LOG  DFU state: DFU_PROCESS_STARTING
>  LOG  DFU state: DFU_PROCESS_STARTING
>  LOG  DFU progress: 0
>  LOG  DFU progress: 0
>  LOG  DFU state: DEVICE_DISCONNECTING
>  LOG  DFU state: DEVICE_DISCONNECTING
>  LOG  DFU state: DFU_FAILED
>  LOG  DFU state: DFU_FAILED
>  LOG  NordicDFU.startDFU [Error: OPERATION FAILED]

I have try setting a low MTU or numberOfRetries (viewing some forks) but not get working.

p4bl1t0, 2022-04-29:
Well same on android ... if anyone get to here i have found that 

```
.setPacketsReceiptNotificationsEnabled(true)
.setPacketsReceiptNotificationsValue(1)
```

askkerr, 2023-11-03:
> Well same on android ... if anyone get to here i have found that
> 
> ```
> .setPacketsReceiptNotificationsEnabled(true)
> .setPacketsReceiptNotificationsValue(1)
> ```

What location did you make this change?