Pilloxa/react-native-nordic-dfu #101: manual installation iOS

tal987, 2020-06-10:
Hi,

There is a way to get the manual installation for iOS? I know it might be hard but I would like to do it manually... can someone supply it?

ezranbayantemur, 2020-06-10:
Check this fork's readme, it may be help: https://github.com/PolideaInternal/react-native-nordic-dfu

tal987, 2020-06-11:
@ezranbayantemur hey, thanks for it, but I didn't succeed to install it... one of the steps saying `Drag and drop Zip.framework and iOSDFULibrary.framework into your project` but I couldn't find those frameworks... where are they located?

ezranbayantemur, 2020-06-11:
@tal987 I made installation by automatic version and got no error. So I didn't do these manual steps. I suggest you to use automatic way because these files maybe changed by the version.

tal987, 2020-06-11:
@ezranbayantemur thanks for the response... I tried the pod installation, but its not working, can you please check out this post with my issue:
https://github.com/Pilloxa/react-native-nordic-dfu/issues/105

ezranbayantemur, 2020-06-11:
Sure, let's check it out