Pilloxa/react-native-nordic-dfu #105: can't see the library after pod install

tal987, 2020-06-11:
hey, so I did exactly like I should install the library with `pod install`
it seems it works and installed but I don't see the the library files on the project...

this is the logs from the pod install:

    ` pod install
      /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/lib/ruby/2.6.0/universal-darwin19/rbconfig.rb:229: warning: Insecure world writable dir /Users/x in PATH, mode 040777
     Analyzing dependencies
     Fetching podspec for react-native-nordic-dfu from ../node_modules/react-native-nordic-dfu
     Downloading dependencies
     Using Firebase (5.7.0)
     Using FirebaseAnalytics (5.1.1)
     Using FirebaseCore (5.1.2)
     Using FirebaseInstanceID (3.2.1)
     Using FirebaseMessaging (3.1.1)
     Using GoogleAppMeasurement (5.1.1)
     Using GoogleUtilities (5.2.2)
     Using Protobuf (3.6.1)
     Installing React (0.11.0)
     Installing ZIPFoundation (0.9.11)
     Installing iOSDFULibrary (4.7.2)
     Using nanopb (0.3.8)
     Installing react-native-nordic-dfu (3.2.0)
     Generating Pods project
     Integrating client project
     Sending stats
     Pod installation complete! There are 4 dependencies from the Podfile and 13 total pods installed.

    [!] The `APP [Debug]` target overrides the `ALWAYS_EMBED_SWIFT_STANDARD_LIBRARIES` build setting defined in `Pods/Target Support Files/Pods-APP/Pods-APP.debug.xcconfig'. This can lead to problems with the CocoaPods installation
    - Use the `$(inherited)` flag, or
    - Remove the build settings from the target.

     [!] The `APP [Release]` target overrides the `ALWAYS_EMBED_SWIFT_STANDARD_LIBRARIES` build setting defined in `Pods/Target Support Files/Pods-Vitalacy/Pods-APP.release.xcconfig'. This can lead to problems with the CocoaPods installation
    - Use the `$(inherited)` flag, or
    - Remove the build settings from the target.

      [!] React has been deprecated`

For example I don't see the `RNNordicDfu.xcodeproj` under library folder

what did I miss?

ezranbayantemur, 2020-06-11:
Looks like the problem is because of project build settings, on iOS side. But we have to be sure about some conditions; 
Can you try to load this library on fresh, new react-native project? If it's run okay on fresh project then looks like your current project is corrupted. 
Also can you run `npx react-native info` on your current project directory?

tal987, 2020-06-11:
@ezranbayantemur after running the npx command I got:
  `System:
    OS: macOS 10.15.4
    CPU: (4) x64 Intel(R) Core(TM) i5-7360U CPU @ 2.30GHz
    Memory: 1.08 GB / 8.00 GB
    Shell: 3.2.57 - /bin/bash
  Binaries:
    Node: 10.0.0 - /usr/local/bin/node
    npm: 6.4.1 - /usr/local/bin/npm
    Watchman: 4.9.0 - /usr/local/bin/watchman
  SDKs:
    iOS SDK:
      Platforms: iOS 13.4, DriverKit 19.0, macOS 10.15, tvOS 13.4, watchOS 6.2
  IDEs:
    Xcode: 11.4.1/11E503a - /usr/bin/xcodebuild
  npmPackages:
    react: ^16.8.6 => 16.8.6 
    react-native: ^0.60.5 => 0.60.5 
  npmGlobalPackages:
    react-native-awesome-alerts: 1.1.1
    react-native-cli: 2.0.1
    react-native-device-info: 2.1.2
    react-native-git-upgrade: 0.2.7`

about the new project ill try it soon and let you know

tal987, 2020-06-11:
@ezranbayantemur about creating a new project and try install it, I didn't get any issue and it's working, what might be my problem?

ezranbayantemur, 2020-06-12:
Yes probably it was it. You didn't get any error on fresh project so problem wasn't because of this library. Probably your project get corrupted because of other packages installation process. Maybe you made changes on XCode. 
If you want you can try it another fresh project which is `0.60.5` version of React Native but I don't think it would be a problem actually.

tal987, 2020-06-12:
@ezranbayantemur do you think reinstall pod on my project can help?

ezranbayantemur, 2020-06-12:
Maybe. You should track the errors

tal987, 2020-06-12:
@ezranbayantemur  the think that I don’t get any error... do you know how can I reset the pod? Or do you have any other option how should I fix it?
Thank you!

ezranbayantemur, 2020-06-12:
Try to build on XCode and track it on there. You'll get infos on output section
If you use you can try Filpper for it too

ezranbayantemur, 2020-06-12:
I had issues on Android and solve it on Logcat, on Studio. Think the same

tal987, 2020-06-15:
@ezranbayantemur I installed the library with pod, but I keep getting an error: 
Showing Recent Messages
node_modules/react-native-nordic-dfu/ios/RNNordicDfu.h:2:9: 'React/RCTBridgeModule.h' file not found...

I even try to open a new project and install it but the build failed on the same error, as I understand it happens because the `use_framework!` but I think is necessary for this library...