Pilloxa/react-native-nordic-dfu #107: Support react-native-ble-plx

andrewcharnley, 2020-06-14:
react-native-ble-manager is quite broken with loads of simple errors which should have caught during testing. Any chance of including support for react-native-ble-plx?

Looveh, 2020-06-14:
This library doesn't depend on `react-native-ble-manager` and should be compatible with `react-native-ble-plx`. 

On iOS the only requirement is that there's a way to retrieve a reference to the instance of `CBCentralManager` used by `react-native-ble-plx` and then pass it to this library. I haven't worked with that library myself but after a quick search it seems to be exposed here: https://github.com/Polidea/MultiPlatformBleAdapter/blob/master/iOS/RxBluetoothKit/BluetoothManager.swift#L63

On Android I can't see any limitations to using `react-native-ble-plx` straight out of the box.

andrewcharnley, 2020-06-16:
Oh right, I've misunderstood the documentation. Excellent then!

closed 2020-06-16 by andrewcharnley

jbreuer95, 2020-06-16:
@andrewcharnley Can you post an example if you manage to get this working with react-native-ble-plx?

andrewcharnley, 2020-06-17:
Will do, I have to get it working... react-native-ble-manager is a POS.

williamgoodhew, 2020-10-16:
@andrewcharnley did you manage to get this working?

andrewcharnley, 2020-11-04:
Sorry for the delay.

Yes, it's working well without any special integration. I fully recommend react-native-ble-plx over react-native-ble-manager.

 As Looveh comments above on Android 'it just works'. I haven't tested on iOS at all but will do a couple months down the line.

jjduhamel, 2021-08-01:
[This issue](https://github.com/dotintent/react-native-ble-plx/issues/260) provides a solution for iOS.

tudormarze, 2021-09-28:
Unfortunately the solution with providing an empty `CBCentralManager` doesn't work, as it's specified also in the documentation 

```
Observe: The peripheral must have been discovered by the native BLE side so that the bluetooth stack knows about it. This library will not do a scan but only the actual connect and then the transfer. See the example project to see how it can be done in React Native.
```

Managed to fix it by writing a native iOS module that is called from React Native. It implements `CBCentralManagerDelegate`, instantiates a `CBCentralManager` and triggers the scan after receiving a `PoweredOn` state update from `centralManagerDidUpdateState`. When the scan finds the device, a callback is passed back to React Native to start the DFU in the library.

tomaseliasm, 2021-12-06:
Hi @tudormarze!
Can you provide an example?

alfredcheuk, 2021-12-22:
Hey @tomaseliasm I jumped through tons of related issues around `react-native-ble-plx` and `react-native-nordic-dfu` workarounds, and somehow put together a way that works. 

Instead of writing a native iOS module as the previous person commented (which is probably the better way), I referred to [this](https://github.com/Pilloxa/react-native-nordic-dfu/issues/78) issue and the first comment. Note that as commenters in there also say, it is not necessarily the best practice. 

In summary these are the steps I took to make ble-plx and nordic-dfu libraries work together on iOS, your mileage may vary:
1. Add the code [here](https://github.com/dotintent/react-native-ble-plx/issues/260#issuecomment-387798491) to my app's ios/app_name/AppDelegate.m file, basically what was instructed in this library's readme page
2. Doing step 1 gave me the error of `Could not find device with deviceAddress`. To work around that I went into my node_modules/react-native-nordic-dfu/ios/RNNordicDfu.m and add the sleep code according to [this comment](https://github.com/Pilloxa/react-native-nordic-dfu/issues/78).

Again I am sure what I did here is not the optimal practice and may have long term issues, but it is the only way I have found from digging through issues and something I was able to work out (I don't have native code knowledge).

tomaseliasm, 2021-12-22:
Hi @alfredcheuk, thanks for the reply!