Pilloxa/react-native-nordic-dfu #108: Allow embedded firmware update data

andrewcharnley, 2020-06-17:
Rather than the user selecting a file I'm intending to include a firmware update with the React App itself. To do this I've base64 encoded the DFU payload into a json file (since ReactNative can't bundle files on IOS). 

As a workaround I might be able to dump the data back out to the App's file space but I'd rather just fire over the payload directly to react-native-nordic-dfu.startDFU.

If anyone is interested in this method here's a bash script I used to do the work and bump the version number.

`#!/usr/bin/bash -e

perl -i -pe 's/\b(\d+)(?=\D*$)/$1+1/e' .APP_VERSION
VERSION=$(< .APP_VERSION)
./bootloader/nrfutil-linux pkg generate --hw-version 52 --application-version $VERSION --application ./app/Output/810-release/Exe/c1.hex --sd-req 0xCD --key-file ./bootloader/private.key /tmp/dfu.zip
base64 -w 0 /tmp/dfu.zip > /tmp/dfu.64
DFU64=$(< /tmp/dfu.64)
echo "{ \"payload\":\"${DFU64}\"}" > ./reactnative/src/assets/dfu.json
`

Looveh, 2020-06-18:
@andrewcharnley Interesting workaround, let us know how it turns out!

andrewcharnley, 2020-06-19:
From what I can tell your library is just interfacing with the nordic library so the easiest method would be to pass over a base64://xxx string to the fileName.

andrewcharnley, 2020-07-10:
This is now on my radar but unfortunately I have zero Java/native development experience.

I can see that the java package is imported at https://github.com/Pilloxa/react-native-nordic-dfu/blob/master/android/build.gradle

It would be best if I could clone https://github.com/NordicSemiconductor/Android-DFU-Library and point the import to it. Any help on how to do this much appreciated!