Pilloxa/react-native-nordic-dfu #119: Android multi part DFU DEVICE DISCONNECTED

RRMoelker, 2020-11-04:
The problem we are facing is that a multi part DFU fails on Android.

After calling `NordicDFU.startDFU` everything runs fine until the first part of a multi part DFU is done. Directly following the 100% upload of the first part the function yields the error: "DFU DEVICE DISCONNECTED"

When interacting with the clock after that bonding is needed again. Which indicates to me that maybe the second upload is failing because it may need to wait until a new bond is needed.

The same update works just fine on iOS.

I'm no expert, but I would think either:
* Allow more granular control over which part is uploaded when.
* Or allow for bonding between parts.

RRMoelker, 2020-11-04:
We've seen this problem on multiple devices running Android 10 and 11. So far Android 9 seems to work fine.

deniske, 2021-01-16:
Hi @RRMoelker ,

      I met the same issue, do you find the solution?

RRMoelker, 2021-01-16:
Yes, although I don't recall the solution.

There are multiple known issues with known fixes. I merge them into a fork of this repo: https://github.com/DreamOnward/react-native-nordic-dfu.

deniske, 2021-01-19:
Hi @RRMoelker ,

Thanks for the answer. But I still met the problems. Did you met the issue of GATT Error when you do DFU upgrade on android phone?

RRMoelker, 2021-01-19:
The GATT error we encountered here were two fold.

1. After a firmware upgrade the Bluetooth service might have changed. These changes aren't always picked up by the OS (iOS & Android). A full rebonding is sometimes needed. On Android you can throw away the old bond and force rebonding. On iOS you need to direct the user to manually remove the existing Bluetooth devices from their devices list.
2. 
2. A slew of newer Samsung devices have a software problem where they don't handle changes to the MTU properly. I think they have fixed their OS. But the bottom line is that many Samsung devices will (Gatt) error on you during DFU. In the fork I mentioned I made it possible to force the MTU to 23 so this does not happen. We do that for all Samsung devices (in lieu of a list of affected devices).

More info on Samsung issue: https://forum.developer.samsung.com/t/samsung-android-10-ble-connectivity-regression/509

hasnainalimj, 2021-11-16:
You can fix this by changing the value of the method `setKeepBond` from `true` into `false` inside [](https://github.com/Pilloxa/react-native-nordic-dfu/blob/master/android/src/main/java/com/pilloxa/dfu/RNNordicDfuModule.java)

`public void startDFU(String address, String name, String filePath, Promise promise) {
        mPromise = promise;
        final DfuServiceInitiator starter = new DfuServiceInitiator(address)
                .setKeepBond(false);`