Pilloxa/react-native-nordic-dfu #121: "Could not find device with deviceAddress" once I start to update the device.

shakogele, 2020-11-11:
I am getting this error : "Could not find device with deviceAddress"
error code: "unable_to_find_device"

when I do 
```
NordicDFU.startDFU({
                            deviceAddress: scannedDevice.id,
                            deviceName: scannedDevice.name,
                            filePath: res.data,
                            alternativeAdvertisingNameEnabled: false,
                        })
```
I am using [react-native-ble-plx ](https://github.com/Polidea/react-native-ble-plx) library and as it does not export BLEManager I used [this approach](https://github.com/Polidea/react-native-ble-plx/issues/260)  to make it work, but now it is not able to discover peripherals. Is there any workaround for this issue?

deniske, 2020-11-16:
Hi shakogele
    Did you find the solution?

mayue3434, 2021-01-21:
The same issue on iPhone 8p, iOS14.3. Any solution?

RupmalyaK, 2021-11-17:
Same on iOS 15

david-gettins, 2022-04-08:
To save people from having to follow various links and collate the information, here is the solution that works for me when using [react-native-ble-plx](https://github.com/dotintent/react-native-ble-plx)...

**1. Your AppDelegate.m**
```objective-c
- (BOOL)application:(UIApplication *)application didFinishLaunchingWithOptions:(NSDictionary *)launchOptions
{
  // ...

  [RNNordicDfu setCentralManagerGetter:^() {
      return [[CBCentralManager alloc] initWithDelegate:nil queue:dispatch_get_global_queue(DISPATCH_QUEUE_PRIORITY_BACKGROUND, 0)];
  }];

  [RNNordicDfu setOnDFUComplete:^() {
  }];

  [RNNordicDfu setOnDFUError:^() {
  }];

  // ...
}
```

Originally from [here](https://github.com/dotintent/react-native-ble-plx/issues/260#issuecomment-387798491)

**2. Inside the react-native-nordic-dfu folder in node_modules - RNNordicDfu.m**

At line 200 or underneath the line: `CBCentralManager * centralManager = getCentralManager();` add this:

```objective-c
[NSThread sleepForTimeInterval: 2];
```

Originally from [here](https://github.com/Pilloxa/react-native-nordic-dfu/issues/78#issue-506247368)

Now everything works as expected. Apologies for duplicating this information but it took me 15mins to bring various bits from other issue comments together, hopefully this saves someone that time.

Finally, here is the patch I generated to apply this on install using [patch-package](https://github.com/ds300/patch-package):
```diff
diff --git a/node_modules/react-native-nordic-dfu/ios/RNNordicDfu.m b/node_modules/react-native-nordic-dfu/ios/RNNordicDfu.m
index ed58530..4c8b4a7 100644
--- a/node_modules/react-native-nordic-dfu/ios/RNNordicDfu.m
+++ b/node_modules/react-native-nordic-dfu/ios/RNNordicDfu.m
@@ -197,6 +197,7 @@ - (void)logWith:(enum LogLevel)level message:(NSString * _Nonnull)message
     reject(@"nil_central_manager_getter", @"Attempted to start DFU without central manager getter", nil);
   } else {
     CBCentralManager * centralManager = getCentralManager();
+    [NSThread sleepForTimeInterval: 2];
 
     if (!centralManager) {
       reject(@"nil_central_manager", @"Call to getCentralManager returned nil", nil);
```