Pilloxa/react-native-nordic-dfu #128: Getting every time DFU DEVICE DISCONNECTED error

leelaprasadsahu22, 2021-03-04:
Getting every time DFU DEVICE DISCONNECTED error 10-20 seconds after calling NordicDFU.startDFU function.
Not getting only state Failed 
DFUEmitter.addListener("DFUStateChanged", (state) => {
                                            console.log("DFU State:", state);
                                        })

AshirbadGudu, 2021-06-03:
### I'm also Getting Same Error 
#### Here I'm showing one video what is the exact issue so that everyone can check

https://user-images.githubusercontent.com/49669234/120633617-f8e02400-c487-11eb-82b7-5c71f9cb7ada.mp4

#### Explaining my code step wise
##### Step-1: Enable Bluetooth Connection via BleManager
```js
  const enableBluetooth = () => {
    BleManager.enableBluetooth()
      .then(() => {     
        Alert.alert('Success', 'Bluetooth enabled successfully');
      })
      .catch((error) => {
        console.log(error);
      });
  }
```
##### Step-2: Connect with sensor by its mac ID via BleManager
```js
const connectWithSensor = () => {
    BleManager.connect(sensorMacId)
        .then(() => {          
          Alert.alert('Success', 'Sensor Connected Successfully');
        })
        .catch((error) => {
          console.log(error);
        });
  }
```
##### Step-3: Select Firmware File and Strat DFU process
```js
const sendFileFromYourDevice = async () => {
    try {
      const firmwareFile = await DocumentPicker.pick({
        type: DocumentPicker.types.zip,
      });
      const destination = RNFS.CachesDirectoryPath + '/firmwareFile.zip';
      await RNFS.copyFile(firmwareFile.uri, destination);
      NordicDFU.startDFU({deviceAddress: sensorMacId, filePath: destination});
    } catch (error) {
      console.log(error);
    }
  }
```

**Please Share Any Solution If Any One Has**

xanderdeseyn, 2022-08-26:
Did you find any solution to this? We are seeing the same thing.

xjoexripx, 2023-05-05:
I was hitting this issue, and a co-worker helped me find that the issue was advertising/connection interval differences.

The firmware I had running had a different advertising/connection interval than the boot loader. The react-native-nordic-dfu would connect to my device successfully, and then put the device into boot loader mode. It would then try to connect, and fail (likely due to BLE advertising/connection setting differences). 

After modifying my firmware's BLE advertising/connection intervals to match the boot loader's, I have had no issues.

It is worth pointing out that the Pixel 2 had no issue DFUing, however the Pixel 7 would not work (until applying the fix above).

It is also worth pointing out that this solve should only apply if you are able to put the device into boot loader mode, but then get the "DFU DEVICE DISCONNECTED" error.