Pilloxa/react-native-nordic-dfu #144: Getting this library to work without react-native-document-picker

jjduhamel, 2021-08-24:
Hello,

I'm running into an issue very similar to [this](https://github.com/Pilloxa/react-native-nordic-dfu/issues/100).  The difference in my case is that react-native-document-picker is stubbornly not working in my project (see [here](https://github.com/rnmods/react-native-document-picker/discussions/461)), so I'm using react-native-fs to derive the file path instead.

I can't find any info about the `public.archive` filetype on iOS, nor was an explanation ever given for the issue I posted earlier.  Assuming that react-native-document-picker won't work for me, can you give me any guidance for making this work with an alternate filesystem library?

Thanks

jjduhamel, 2021-08-24:
Here is the actual error I'm getting:

```
Exception '*** -[__NSPlaceholderArray initWithObjects:count:]: attempt to insert nil object from objects[0]' was thrown while invoking startDFU on target RNNordicDfu with params (
    "D6:F2:09:33:AE:AE",
    EuphoriaLight,
    "/var/mobile/Containers/Data/Application/A572D708-0143-4277-9B8D-62681B194336/Documents/bundle_euphoria.zip",
    1,
    896,
    897
)
callstack: (
	0   CoreFoundation                      0x000000018990271c 338F58B1-9B75-38A0-B908-E2574AB1D39A + 1218332
	1   libobjc.A.dylib                     0x000000019e40c7a8 objc_exception_throw + 60
	2   CoreFoundation                      0x00000001899749c8 338F58B1-9B75-38A0-B908-E2574AB1D39A + 1685960
	3   CoreFoundation                      0x00000001899723f4 338F58B1-9B75-38A0-B908-E2574AB1D39A + 1676276
	4   CoreFoundation                      0x00000001897e39d0 338F58B1-9B75-38A0-B908-E2574AB1D39A + 43472
	5   CoreFoundation                      0x00000001897de484 338F58B1-9B75-38A0-B908-E2574AB1D39A + 21636
	6   EuphoriaApp                         0x0000000100aaa620 -[RNNordicDfu startDFU:deviceName:filePath:alternativeAdvertisingNameEnabled:resolver:rejecter:] + 532
	7   CoreFoundation                      0x0000000189907724 338F58B1-9B75-38A0-B908-E2574AB1D39A + 1238820
	8   CoreFoundation                      0x00000001897dad04 338F58B1-9B75-38A0-B908-E2574AB1D39A + 7428
	9   CoreFoundation                      0x00000001897db308 338F58B1-9B75-38A0-B908-E2574AB1D39A + 8968
	10  EuphoriaApp                         0x00000001007d01fc -[RCTModuleMethod invokeWithBridge:module:arguments:] + 1752
	11  EuphoriaApp                         0x00000001007d3540 _ZN8facebook5reactL11invokeInnerEP9RCTBridgeP13RCTModuleDatajRKN5folly7dynamicE + 604
	12  EuphoriaApp                         0x00000001007d3120 _ZZN8facebook5react15RCTNativeModule6invokeEjON5folly7dynamicEiENK3$_0clEv + 128
	13  EuphoriaApp                         0x00000001007d3094 ___ZN8facebook5react15RCTNativeModule6invokeEjON5folly7dynamicEi_block_invoke + 28
	14  libdispatch.dylib                   0x00000001894eda84 9555D7D9-9C92-3AF3-9837-AE1F5AA7DA44 + 10884
	15  libdispatch.dylib                   0x00000001894ef81c 9555D7D9-9C92-3AF3-9837-AE1F5AA7DA44 + 18460
	16  libdispatch.dylib                   0x00000001894f7004 9555D7D9-9C92-3AF3-9837-AE1F5AA7DA44 + 49156
	17  libdispatch.dylib                   0x00000001894f7c00 9555D7D9-9C92-3AF3-9837-AE1F5AA7DA44 + 52224
	18  libdispatch.dylib                   0x00000001895024bc 9555D7D9-9C92-3AF3-9837-AE1F5AA7DA44 + 95420
	19  libsystem_pthread.dylib             0x00000001d568e7a4 _pthread_wqthread + 276
	20  libsystem_pthread.dylib             0x00000001d569574c start_wqthread + 8
)
```

tomaseliasm, 2021-12-09:
Hi @jjduhamel did you managed to get this working with react-native-fs?