Pilloxa/react-native-nordic-dfu #155: Build failed on Xcode 13

suryakant-invitra, 2021-11-24:
Environment:
Xcode : 13
iOS SDK simulator : 15
React-native : 0.63.3
Mac OSX : Monterey
[image]

Xcode 12 everything was working fine but after Xcode upgrade throwing above errors.
Please help me to resolve this issue.

kgaspar, 2021-12-01:
Anyone got a solution for this? I've been getting an error with the underlying Multiplatform adapter. Think It might be related to the new Swift additions in XCode 13

suryakant-invitra, 2021-12-02:
@kgaspar, I have resolved the above error by running `pod update iOSDFULibrary` from ios directory.

kgaspar, 2021-12-03:
Unfortunately, the error persists even after using the two commands. Anyone had any luck?

kgaspar, 2021-12-03:
I managed to resolve this issue by updating to the new version of the library 3.2.1 and doing the above steps. Thanks for the help and thats to the maintainers for the update.