Pilloxa/react-native-nordic-dfu #160: Could not find device with deviceAddress [3.2.1]

marcopiraccini, 2022-03-01:
I am having this issue when I `startDFU` on iOS (on Android it works). 
I already checked and the `deviceAddress` is the UUID of the device (at it should be).
Any idea?

marcopiraccini, 2022-03-01:
I guess the error comes from here: https://github.com/Pilloxa/react-native-nordic-dfu/blob/master/ios/RNNordicDfu.m#L210

marcopiraccini, 2022-03-01:
OK, this is probably a duplicate of: https://github.com/Pilloxa/react-native-nordic-dfu/issues/121 (I am also using [react-native-ble-plx ](https://github.com/Polidea/react-native-ble-plx))

marcopiraccini, 2022-04-11:
OK, but there is a reason why this patch is not applied directly to the code?

p4bl1t0, 2022-04-20:
Yes, this should be a Pull Request or be added to the library

david-gettins, 2022-05-27:
@marcopiraccini Did you get around to preparing the PR?

Irenedan, 2022-09-20:
@marcopiraccini, I'm using the "react-native-ble-plx" too, but I don't know what parts are zip made of. Could you please put a zip file of yours here?

`NordicDFU.startDFU({
  deviceAddress: '94:54:93:17:63:75', 
  deviceName: 'test', 
  filePath: fs.CachesDirectoryPath + '/dry.zip'
})`

p4bl1t0, 2022-09-20:
The zip file should be your firmware code compiled by Arduino or Platform.io with the upload_protocol in nrfutil