Pilloxa/react-native-nordic-dfu #166: Android app crashes at startDfu when targeting SDK version 31+

alfredcheuk, 2022-05-26:
When running startDfu function on Android while targeting SDK version 31, the app crashes directly without any errors thrown. 

I used adb logcat to trace the log messages and found the following issue:
[image]
(masked out the project name that I am working on, and the first line was me printing the firmware file for debugging)

I am not the most familiar with logcat outputs but it seems that it ran into the exception on the highlighted line, then the app proceeded to crash directly.

I have tested that the exact same code works on iOS, as well as Android while targeting SDK version 30.

Does anyone have any idea whether this is something that needs to be changed on the library's side, or have I not configured it correctly specifically for SDK version 31?

vkourtis, 2022-06-09:
There is and open PR that fixes that https://github.com/Pilloxa/react-native-nordic-dfu/pull/164

alfredcheuk, 2022-06-09:
Thank you so much, I will keep an eye out on this.

vkourtis, 2022-06-09:
> Thank you so much, I will keep an eye out on this. 

Check [my comment](https://github.com/Pilloxa/react-native-nordic-dfu/pull/164#issuecomment-1150974953) on the PR. I was able to patch it with `yarn patch`.

Cantinaband, 2022-06-15:
Having the same issue with Android 12. Would be great to have this PR released

fabiendem, 2022-06-17:
Is there any maintainer left on this library? Would be good to merge this.

A `targetSdkVersion` set to Android 12 (API 31) will be mandatory in November 2022: https://support.google.com/googleplay/android-developer/answer/11926878

fabiendem, 2022-08-18:
Fix available in this fork: https://github.com/Salt-PepperEngineering/react-native-nordic-dfu

sachinsaini794, 2022-09-22:
> 

> vkou

Yes it's working now on android 12. The node_modules was not updating the code. The package.json file shows the 3.2.1 version but it's code was not added in the node_modules folder so I have created the patch package and replace the android folder . Now it's working fine.

kakonyitibor, 2022-11-22:
Hello, 
I had the same issue, the app crashed on startDFU with Android12.
I'm trying PR #164 and although it doesn't crash anymore, but process gets stuck in the DFU_PROCESS_STARTING state, then throws GATT ERROR. (I don't have this problem with older ANdroid versions. )
Can you help me please?
Thank you!

sachinsaini794, 2022-11-24:
@kakonyitibor  have you used the patch for the react-native-nordic-dfu? if not please use it and update the code as in package 3.2.1.

for example - 
+def DEFAULT_COMPILE_SDK_VERSION = 31
+def DEFAULT_BUILD_TOOLS_VERSION = '31.0.0'
+def DEFAULT_TARGET_SDK_VERSION = 31

kakonyitibor, 2022-11-24:
> @kakonyitibor have you used the patch for the react-native-nordic-dfu? if not please use it and update the code as in package 3.2.1.
> 
> compare with this https://github.com/Salt-PepperEngineering/react-native-nordic-dfu/blob/master/android/build.gradle
> 
> for example - +def DEFAULT_COMPILE_SDK_VERSION = 31 +def DEFAULT_BUILD_TOOLS_VERSION = '31.0.0' +def DEFAULT_TARGET_SDK_VERSION = 31

Thanks for your reply. 
Yes, I'm using the patch, build.gradle is the same as the one you linked. Thanks to this patch, DFU does not crash the app on Android 12 anymore. Now it just throws a GATT ERROR right after starting the DFU.