Pilloxa/react-native-nordic-dfu #167: [Question] Is this library still maintained ?

manualexSP, 2022-05-31:
Hello everyone,

I see that are no new releases from 2020 and since then there are a couple of issues that appeared with the new ios/android versions. This library is very important for the project that I’m currently working so I created a fork and fixed some of the issues.

My question is, is this library still maintained? because I created this [PR](https://github.com/Pilloxa/react-native-nordic-dfu/pull/164) which could help others as well. Is there any maintainer available to review and release it?

Thank you.

manualexSP, 2022-06-21:
I already have a fork that I'm maintaining. I just wanted to know if I come in the future with some PR, someone can review them.
But I already see someone used some code from my previous [PR](https://github.com/Pilloxa/react-native-nordic-dfu/pull/164) so this issue can be closed.

closed 2022-06-21 by manualexSP

fabiendem, 2022-06-21:
@yannvanhalewyn Thanks for getting back.
Totally understand and thanks for the clarifications.

@manualexSP Do you plan to merge https://github.com/Pilloxa/react-native-nordic-dfu/pull/164 on your own fork?

manualexSP, 2022-06-21:
@fabiendem  is already merged [there](https://github.com/Salt-PepperEngineering/react-native-nordic-dfu) feel free to try it out. 
@yannvanhalewyn of course if this helps others.

yannvanhalewyn, 2022-06-21:
@manualexSP great, I have updated the README with a warning and a link to your fork. Thanks!

yannvanhalewyn, 2022-06-23:
@manualexSP would you like to have a call to sync up about this? Shouldn't take too long. We can do via clojurians.slack.com (I'm yannvanhalewyn there) or another route if you prefer.

manualexSP, 2022-06-24:
@yannvanhalewyn I will sent you today an email an we can disscuss there more details.