Pilloxa/react-native-nordic-dfu #169: BEWARE of the linked fork in the readme

CptFabulouso, 2022-06-29:
If you do not target Android 12 and your app is running on Android 12, the suggest fork [Salt-PepperEngineering/react-native-nordic-dfu](https://github.com/Salt-PepperEngineering/react-native-nordic-dfu) will somehow remove BLUETOOTH_CONNECT and BLUETOOTH_SCAN permissions (which are granted automatically if you do not target Android 12), leading to your app crashing whenever using Bluetooth.

manualexSP, 2022-07-07:
Hi @CptFabulouso, on my project, I'm not targeting android 12 and I don't have this problem. 
On build.gradle  I have :
```
       minSdkVersion = 21
       compileSdkVersion = 30
       targetSdkVersion = 30
```
And we tested on several android 12 devices and we didn't have any problems.
The fix for the android 12 crash issue was to update the [Android-DFU-Library](https://github.com/NordicSemiconductor/Android-DFU-Library) from 1.8.0 to 2.0.1 and the target SDK to 31.

Could you please give me more information about your problem maybe I missed something?

manualexSP, 2022-07-11:
@CptFabulouso ok so we have the same gradle and react-native version, and I'm using the ble-plx for Bluetooth.

Can you please try to declare the permissions like this:
```
    <uses-permission
        android:name="android.permission.BLUETOOTH"
        android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />
    <!-- API 31+ -->
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADVERTISE" />

    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
```
If this is not working then I will set the SDK to 30 and test to see how it works, it shouldn't be a problem because it was just an extra thing that I did.

CptFabulouso, 2022-11-29:
@manualexSP Hi, I finally got around to test this once again with clean RN project to eliminate issues in my setup. I created this repo to test this out https://github.com/CptFabulouso/nordic-dfu-fork-issue-repo

There's readme how to simulate the crash, look for `my-case`, which is what happened in my project. I basically had some old setup in manifest with
```
<uses-permission android:name="android.permission.BLUETOOTH"/>
<uses-permission android:name="android.permission.BLUETOOTH_ADMIN"/>
<uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
<uses-permission android:name="android.permission.FOREGROUND_SERVICE" />
<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
```
and with that the app was working on Android 12 fine. After using the fork app crashes and it is required to specify and use `BLUETOOTH_CONNECT(SCAN)` even without changing the `targetSdkVersion`. You can look at `Case fix` commit in `my-case` branch.