Pilloxa/react-native-nordic-dfu #181: Progress at 100% but I don't get DFU_COMPLETED

jorgefreitaslh, 2023-02-15:
Update firmware started 
`DFU_PROCESS_STARTING` 

Progress at 100% 

But I don't get 
`DFU_COMPLETED`

p4bl1t0, 2023-02-24:
Experience the same issue. At logcat seeing 

02-24 17:34:34.237  2515  2627 D BtGatt.GattService: onClientConnUpdate() - connId=8, status=0
02-24 17:34:34.238 16859 17322 D BluetoothGatt: onConnectionUpdated() - Device=EF:8A:77:AD:DA:29 interval=39 latency=0 timeout=500 status=0
02-24 17:34:35.204 16859 17423 I DfuImpl : Reading DFU version number...
02-24 17:34:35.279  2515  2902 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 11, msg_len = 2
02-24 17:34:35.284 16859 17423 I DfuImpl : Version number read: 0.8 -> Bootloader from SDK 9.0 or newer. Signature supported
02-24 17:34:35.287 16859 17423 W DfuImpl : Legacy DFU bootloader found
02-24 17:34:35.288 16859 17423 I DfuImpl : Requesting MTU = 517
02-24 17:34:35.288 16859 17423 D BluetoothGatt: configureMTU() - device: EF:8A:77:AD:DA:29 mtu: 517
02-24 17:34:35.289  2515  2532 D BtGatt.GattService: configureMTU() - address=EF:8A:77:AD:DA:29 mtu=517
02-24 17:34:35.376  2515  2902 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 3, msg_len = 2
02-24 17:34:35.376  2515  2902 W bt_l2cap: l2cble_set_fixed_channel_tx_data_length, request not supported
02-24 17:34:35.376  2515  2627 D BtGatt.GattService: onConfigureMTU() address=EF:8A:77:AD:DA:29, status=0, mtu=247
02-24 17:34:35.377 16859 17322 D BluetoothGatt: onConfigureMTU() - Device=EF:8A:77:AD:DA:29 mtu=247 status=0
02-24 17:34:35.378 16859 17322 I DfuImpl : MTU changed to: 247
02-24 17:34:36.379 16859 17423 I DfuImpl : Enabling notifications...
02-24 17:34:36.380 16859 17423 D BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
02-24 17:34:36.382  2515  2532 D BtGatt.GattService: registerForNotification() - address=EF:8A:77:AD:DA:29 enable: true
02-24 17:34:36.383  2515  2627 D BtGatt.GattService: onRegisterForNotifications() - address=null, status=0, registered=1, handle=18
02-24 17:34:36.448  2515  2902 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 19, msg_len = 0
02-24 17:34:37.452 16859 17423 I DfuImpl : Sending Start DFU command (Op Code = 1, Upload Mode = 4)
02-24 17:34:37.521  2515  2902 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 19, msg_len = 0
02-24 17:34:37.524 16859 17423 I DfuImpl : Sending image size array to DFU Packet (0b, 0b, 172056b)
02-24 17:34:38.204  2515  2902 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 27, msg_len = 5
02-24 17:34:38.204  2515  2902 E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation
02-24 17:34:38.207 16859 17423 I DfuImpl : Sending the Initialize DFU Parameters START (Op Code = 2, Value = 0)
02-24 17:34:38.301  2515  2902 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 19, msg_len = 0
02-24 17:34:38.304 16859 17423 I DfuImpl : Sending 14 bytes of init packet
02-24 17:34:38.305 16859 17423 I DfuImpl : Sending init packet (Value = 52-00-FF-FF-FF-FF-FF-FF-01-00-B7-00-A7-A6)
02-24 17:34:38.310 16859 17423 I DfuImpl : Sending the Initialize DFU Parameters COMPLETE (Op Code = 2, Value = 1)
02-24 17:34:38.398  2515  2902 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 19, msg_len = 0
02-24 17:34:38.399  2515  2902 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 27, msg_len = 5
02-24 17:34:38.399  2515  2902 E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation
02-24 17:34:38.403 16859 17423 I DfuImpl : Sending Receive Firmware Image request (Op Code = 3)
02-24 17:34:38.497  2515  2902 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 19, msg_len = 0
02-24 17:34:38.504 16859 16919 I ReactNativeJS: 'DFU progress:', 0
02-24 17:34:38.516 16859 17423 I DfuImpl : Uploading firmware...
02-24 17:34:38.570  2515  2902 E bt_stack: [ERROR:gatt_cl.cc(225)] gatt_act_write() failed op_code=0x52 rt=143
02-24 17:34:38.579 16859 16919 I ReactNativeJS: 'DFU progress:', 1
02-24 17:34:38.628 16859 16919 I ReactNativeJS: 'DFU progress:', 2
02-24 17:34:38.666 16859 16919 I ReactNativeJS: 'DFU progress:', 3
02-24 17:34:39.198 16859 16919 I ReactNativeJS: 'DFU progress:', 4
02-24 17:34:39.236  2515  2902 E bt_stack: [ERROR:gatt_cl.cc(225)] gatt_act_write() failed op_code=0x52 rt=143
02-24 17:34:39.284 16859 16919 I ReactNativeJS: 'DFU progress:', 5
02-24 17:34:39.934 16859 16919 I ReactNativeJS: 'DFU progress:', 6
02-24 17:34:39.966  2515  2902 E bt_stack: [ERROR:gatt_cl.cc(225)] gatt_act_write() failed op_code=0x52 rt=143
02-24 17:34:40.648 16859 16919 I ReactNativeJS: 'DFU progress:', 7
02-24 17:34:40.686  2515  2902 E bt_stack: [ERROR:gatt_cl.cc(225)] gatt_act_write() failed op_code=0x52 rt=143
02-24 17:34:40.727 16859 16919 I ReactNativeJS: 'DFU progress:', 8
02-24 17:34:41.297 16859 16919 I ReactNativeJS: 'DFU progress:', 9
02-24 17:34:41.321  2515  2902 E bt_stack: [ERROR:gatt_cl.cc(225)] gatt_act_write() failed op_code=0x52 rt=143
02-24 17:34:41.357  1731  1731 W adbd    : timeout expired while flushing socket, closing
02-24 17:34:41.367 16859 16919 I ReactNativeJS: 'DFU progress:', 10
02-24 17:34:41.935 16859 16919 I ReactNativeJS: 'DFU progress:', 11
02-24 17:34:41.970  2515  2902 E bt_stack: [ERROR:gatt_cl.cc(225)] gatt_act_write() failed op_code=0x52 rt=143
02-24 17:34:42.652 16859 16919 I ReactNativeJS: 'DFU progress:', 12
02-24 17:34:42.689  2515  2902 E bt_stack: [ERROR:gatt_cl.cc(225)] gatt_act_write() failed op_code=0x52 rt=143
02-24 17:34:42.736 16859 16919 I ReactNativeJS: 'DFU progress:', 13
[... cut ...]
02-24 17:35:18.963 16859 16919 I ReactNativeJS: 'DFU progress:', 96
02-24 17:35:19.617 16859 16919 I ReactNativeJS: 'DFU progress:', 97
02-24 17:35:19.648  2515  2902 E bt_stack: [ERROR:gatt_cl.cc(225)] gatt_act_write() failed op_code=0x52 rt=143
02-24 17:35:20.239 16859 16919 I ReactNativeJS: 'DFU progress:', 98
02-24 17:35:20.274  2515  2902 E bt_stack: [ERROR:gatt_cl.cc(225)] gatt_act_write() failed op_code=0x52 rt=143
02-24 17:35:20.324 16859 16919 I ReactNativeJS: 'DFU progress:', 99
02-24 17:35:21.029 16859 16919 I ReactNativeJS: 'DFU progress:', 100

jorgefreitaslh, 2023-02-27:
try using 
`starter.setMtu(23);`

in file `RNNordicDfuModule.java`

p4bl1t0, 2023-03-07:
Thnak @jorgefreitaslh works for me.