Pilloxa/react-native-nordic-dfu #182: Firmware File not found - iOS

suryakant-patil, 2023-03-03:
HI,
I am getting the error after starting the DFU process for firmware file not found. 
I have confirmed the firmware file is attached to the process.
Please help me to resolve above issue on iOS

React Native version: 0.63.3

Nickolans, 2023-04-05:
@suryakant-patil Did you find a solution to this?

suryakant-patil, 2023-04-10:
@Nickolans, not yet. Sometimes it works, and sometimes it does not.

Devojha408, 2023-09-12:
@suryakant-patil  try like this:
    const destination = `${RNFS.DocumentDirectoryPath}/installationFile.zip`;
    const exists = await RNFS.exists(destination);
    exists && (await RNFS.unlink(destination));
    const response = await ReactNativeBlobUtil.config({
      fileCache: true,
      appendExt: 'zip',
    }).fetch('GET', firmwareUrl);

    const downloadPath = response.path();
    await RNFS.copyFile(downloadPath, destination);
    response?.flush();
    
     NordicDFU.startDFU({
      deviceAddress: device.id,
      name: device.advertising.localName,
      filePath:
        Platform.OS === 'ios'
          ? `file://${destination}`
          : destination
          });