Pilloxa/react-native-nordic-dfu #185: Android 12+ - README if you get a crash, or try the fork.

andrewcharnley, 2023-07-05:
Hi all,

After many hours....

1. the fork enables jetifier, and I got argument mismatch on the function call that I couldn't resolve. I made the modifications to support v31 and Androidx manually to the non-fork code.

2. the no.nordic.dfu library hard crashes unless you get permissions spot on. I think they have them incorrect in the manifest, or the ble lib you're using does. The best approach is to completely ignore every sodding bluetooth permission in all the sodding manifests and force your own in the master manifest.  After much trial and error these are great for my use case (ble only, no location).

Finally DFU is working again!

`<uses-permission android:name="android.permission.INTERNET" />

    
    <uses-permission android:name="android.permission.BLUETOOTH" android:usesPermissionFlags="neverForLocation" android:maxSdkVersion="31" tools:replace="android:maxSdkVersion" tools:remove="android:minSdkVersion" />
    
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:usesPermissionFlags="neverForLocation" android:maxSdkVersion="30" tools:replace="android:maxSdkVersion" tools:remove="android:minSdkVersion" />

    
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" android:maxSdkVersion="28" tools:replace="android:maxSdkVersion" tools:remove="android:minSdkVersion" />
    
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:minSdkVersion="29" android:maxSdkVersion="30" tools:replace="android:maxSdkVersion,android:minSdkVersion" />

    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation" tools:remove="android:maxSdkVersion,android:minSdkVersion" />

    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" android:usesPermissionFlags="neverForLocation" tools:remove="android:maxSdkVersion,android:minSdkVersion" />
    <uses-permission android:name="android.permission.FOREGROUND_SERVICE" tools:remove="android:maxSdkVersion,android:minSdkVersion" />

    <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>
    
    ...
    
`

And then you need all the runtime permissions as well...

      const permissions = [
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_SCAN,
        PermissionsAndroid.PERMISSIONS.BLUETOOTH_CONNECT
      ];
      // shouldn't be needed on 31 but some manufacturers still require
      if (Platform.Version <= 31) {
        permissions.push(PermissionsAndroid.PERMISSIONS.BLUETOOTH);
      }
      if (Platform.Version <= 30) {
        permissions.push(PermissionsAndroid.PERMISSIONS.BLUETOOTH_ADMIN);
      }
      if (Platform.Version <= 28) {
        permissions.push(PermissionsAndroid.PERMISSIONS.ACCESS_COURSE_LOCATION);
      } else if ([29.30].includes(Platform.Version)) {
        permissions.push(PermissionsAndroid.PERMISSIONS.ACCESS_FINE_LOCATION);
      }
      await PermissionsAndroid.requestMultiple(permissions);
    }

manualexSP, 2023-07-07:
@andrewcharnley  Hi, I'm from the README fork, I just created an [new example](https://github.com/Salt-PepperEngineering/react-native-nordic-dfu/tree/master/newExample) based on a new react-native project. And is included in the setup as well. Let me know if I can add anything in there :D

andrewcharnley, 2023-07-07:
Just remove the useJetifier argument I think, this rewrote the function and I got argument mismatch error - it took me all day to work it out! It may not be obvious until you clear metro cache.