Pilloxa/react-native-nordic-dfu #187: iosdfulibrary/iosdfulibrary-swift.h' file not found

dev-simpleplan, 2023-10-31:
[image]

I am facing some issues related to the IOSDFU library. How can we fix this issue?

andresev, 2023-11-10:
@dev-simpleplan  Any luck on this? I'm getting the same error. I can't really find the ...-Swift.h file anywhere.

dev-simpleplan, 2023-12-13:
> @dev-simpleplan Any luck on this? I'm getting the same error. I can't really find the ...-Swift.h file anywhere.

Yes, I have solved the issue. let me know if you are still facing the issue.

Ritesh9811sahani, 2024-08-27:
[image]

This issue is still coming